raspberrypi/rpi-imager #911: [BUG]: Boot partition not set on PI v1 install

dballanc, 2024-08-24:
### What happened?

Old PI B+, 8gb sdcard.    New image would not boot or display output of any kind.   Checked partitions on card and noticed no partition set bootable.  Added boot flag to partition 1 (512Mb Fat32) and was able to boot successfully.

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

EndeavourOS

### Selected OS

RaspberryPI OS Lite 32bit

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-08-28:
I suspect the "boot flag" might be a red herring, as AFAIK the Raspberry Pi doesn't even use those flags?
Are you able to reproduce this issue on a different SD card?

tdewey-rpi, 2024-08-28:
Thanks for the report, @dballanc 

The Raspberry Pi bootloader does not use the 'bootable' flag in any way, so your resolution was caused by a side-effect of your actions.

I suspect you may have simply not waited long enough for Raspberry Pi 1 to resize the root partition of Raspberry Pi OS. Depending on the capacity and quality of the SDcard, this can take an extraordinarily long time - long enough that in my testing I have wondered if the device had simply failed to boot.

I'm minded to close this, as this isn't an issue with Raspberry Pi Imager.

dballanc, 2024-08-28:
Please close this issue.  I attempted to replicate the problem with a different SDcard and was unable to do so.  I'm not sure why toggling the boot flag worked on the original sd, probably just coincidence.  I don't think it ever got to the resizing stage, the leds on the pi were solid and it acted as if no sdcard were present.  Apologies for the false report.

tdewey-rpi, 2024-08-28:
Not a problem, @dballanc 

Closing at OPs request.

closed 2024-08-28 by tdewey-rpi