raspberrypi/rpi-imager #912: [FEATURE]: macOS native arm64 support

robd003, 2024-08-25:
### Is your feature request related to a problem? Please describe.

The prior release of rpi-imager was released as a Universal Binary which has native support for both Intel & Apple Silicon.

The current release only supports Intel and requires Apple's Rosetta to run which has a detrimental effect when running on a laptop using battery power.

### Describe the solution you would like to see implemented

Release a Universal Binary for rpi-imager in future releases

### Describe alternatives you've considered

Right now the only solution is to use the prior release 1.8.4

### Additional context

Apple will probably deprecate Rosetta support after macOS 15.

Apple only supported the PPC -> Intel Rosetta translation for 2 releases.

### Version

1.8.5 (Default)

tdewey-rpi, 2024-08-28:
Thanks for the report, @robd003 

How have you determined that v1.8.5 isn't a Universal Binary? Your issue piqued my curiosity, so I loaded v1.8.5, and Activity Monitor is indeed showing it as an "Apple" process, rather than an "Intel" one.

tdewey-rpi, 2024-08-30:
Closing as completed - I'm still unclear on how the OP concluded we were not creating Universal Binaries (the source code mandates that at current HEAD), but having confirmed on my local developer machine that 1.8.5 is indeed shipping a Universal binary, I'm going to declare this one complete.

closed 2024-08-30 by tdewey-rpi