raspberrypi/rpi-imager #913: [FEATURE]: User plea - please allow user to specify partition size

frankozland, 2024-08-30:
### Is your feature request related to a problem? Please describe.

many instances i get a default sd card 32g 64g etc.

Raspberry os only takes up 4g.

### Describe the solution you would like to see implemented

Offer advanced parameters for users to alter cmdline.txt,  partition size, modprobe (ie tcp_bbr)

### Describe alternatives you've considered

manual card build

### Additional context

this will solve some advanced user problems and make adoption wider.

### Version

1.8.5 (Default)

lurch, 2024-08-30:
Raspberry Pi OS automatically resizes the root partition (the second one on the SD card) to take up all available space, when you boot it up for the first time :slightly_smiling_face:  

@nathan-contino I wonder if it might be worth mentioning this initial-resize somewhere on https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-the-operating-system ?

tdewey-rpi, 2024-08-30:
Feature request rejected

Raspberry Pi OS - and other operating systems - all perform an online resize of the root partition, obviating the need for Raspberry Pi Imager to understand partitioning beyond its current capability. Expanding that capability will require potentially significant work - work that I am not willing to commit to until we see a problem that cannot be otherwise and more simply worked around.

closed 2024-08-30 by tdewey-rpi