raspberrypi/rpi-imager #914: [BUG]: Pi Zero W Bullseye Lite img, autologin not enabled

xrvk, 2024-09-03:
### What happened?

Hi there,

Not sure if this is intended, but when flashing a new image of a Pi Zero W Bullseye OS Lite, it seems that the autologin is not configured..

I feel it should have autologin enabled by default, especially for the Pi Zero W since I would think most if not all folks wish to run it headless due to lack of ports. Having to plug it into a monitor to manually login locally is quite a hurdle, instead of being able to ssh in right after booting for the first time. 

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 11

### Selected OS

Raspberry Pi Bulleye Lite

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero, Zero W, Zero WH

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-09-04:
Thanks for the report, @xrvk 

Unfortunately what you're asking for isn't clear. I've never seen a lite image autologin - I've only ever seen them boot to a login prompt (which is pretty fair, given their typical use cases).

Your report included a line that suggested you thought Imager, rather than the OS was at fault:

> Having to plug it into a monitor to manually login locally is quite a hurdle, instead of being able to ssh in right after booting for the first time.

Did you fail to connect over SSH after setting up SSH in Imager?

lurch, 2024-09-05:
@xrvk Just to clarify: autologin **isn't** required in order to be able to login remotely over SSH.

xrvk, 2024-09-06:
Oh interesting, sorry for the misunderstanding.

Not sure what happened then... The pi zero w would not connect to wifi after a fresh image install until I manually logged in for the first time and enabled autologin soon after. 

Anyways, if this was just a bug on my end or intended, the feel free to close. I just thought it was odd that I had to manually login on the pi zero w but I never had to do this on the previous Pi 4's I've set up with ssh enabled.

xrvk, 2024-09-09:
I didn't try bookworm since Pihole documentation stated bookworm wasnt supported yet. I'd test it out but I dont have a spare SD card on hand at the moment to test.