raspberrypi/rpi-imager #915: [FEATURE]: Request to Create a Separate Version for PR https://github.com/raspberrypi/rpi-imager/pull/879

ykla, 2024-09-06:
### Is your feature request related to a problem? Please describe.

Would it be possible to create a separate version specifically for the changes in https://github.com/raspberrypi/rpi-imager/pull/879? Your consideration is greatly appreciated. Thank you! I noticed that there hasn't been much activity recently, and it's been 8 months since the last release. Would it be possible to issue an interim release that includes my PR? Thank you for your consideration!

### Describe the solution you would like to see implemented

Create a Separate Version for https://github.com/raspberrypi/rpi-imager/pull/879.

### Describe alternatives you've considered

No.

### Additional context

No.

### Version

1.8.5 (Default)

tdewey-rpi, 2024-09-09:
Thanks for the report, @ykla.

My attentions have indeed been elsewhere for the last few months (Connect, rpi-sb-provisioner, pi-gen, pi-gen-micro), however I am now returning to `rpi-imager` for a new release.

Your PR will be included in the upcoming 1.9.0.

tdewey-rpi, 2024-09-11:
Included in v1.9.0. Let me know if it comes through OK!

closed 2024-09-11 by tdewey-rpi

ykla, 2024-09-12:
Thank you very much for your efforts. I found a relatively obvious but non-critical error in my translation and made corrections to some minor mistakes. I believe this issue doesn't have a significant impact, so there's no need to rush the fix. It can be included in your next release.

ykla, 2024-09-27:
Hi, I find that the <https://www.raspberrypi.com/software/> still releases 1.8.5. Please call them update it. Thanks.