raspberrypi/rpi-imager #916: [BUG]: /etc/wpa_supplicant/wpa_supplicant.conf not getting configured?

jinxin0, 2024-09-08:
### What happened?

Assuming firstrun.sh is what configures the wifi for the pi.
The following snippet of firstrun.sh is outputted to the terminal by rpi-imager when doing my custom config.
```sh
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'wifi_name' 'password' 'US'
else
   cat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'
   country=US
   ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
   ap_scan=1

   update_config=1
   network={
           ssid=\"wifi_name\"
           psk=password
   }
   WPAEOF

   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
   rfkill unblock wifi
   for filename in /var/lib/systemd/rfkill/*:wlan ; do
       echo 0 > $filename
   done
fi
```
If /usr/lib/raspberrypi-sys-mods/imager_custom exist then /etc/wpa_supplicant/wpa_supplicant.conf will not be setup here and the following code will be executed.
```sh
set_wlan () (
  HIDDEN="false"
  PLAIN=0
  for arg in "$@"; do
    # shellcheck disable=SC2031
    if [ "$arg" = "-h" ] || [ "$arg" = "--hidden" ]; then
      HIDDEN="true"
    elif [ "$arg" = "-p" ] || [ "$arg" = "--plain" ]; then
      PLAIN=1
    elif [ -z "${SSID+set}" ]; then
      SSID="$arg"
    elif [ -z "${PASS+set}" ]; then
      PASS="$arg"
    elif [ -z "${COUNTRY+set}" ]; then
      COUNTRY="$arg"
    else
      usage set_wlan
      exit 1
    fi
  done
  if [ -z "${SSID+set}" ]; then
    usage set_wlan
    exit 1
  fi

  if [ -n "$COUNTRY" ]; then
    set_wlan_country "$COUNTRY"
  fi

  CONNFILE=/etc/NetworkManager/system-connections/preconfigured.nmconnection
  UUID=$(uuid -v4)
  cat <<- EOF >${CONNFILE}
	[connection]
	id=preconfigured
	uuid=${UUID}
	type=wifi
	[wifi]
	mode=infrastructure
	ssid=${SSID}
	hidden=${HIDDEN}
	[ipv4]
	method=auto
	[ipv6]
	addr-gen-mode=default
	method=auto
	[proxy]
	EOF

  if [ ! -z "${PASS}" ]; then
    cat <<- EOF >>${CONNFILE}
	[wifi-security]
	key-mgmt=wpa-psk
	psk=${PASS}
	EOF
  fi

  # NetworkManager will ignore nmconnection files with incorrect permissions,
  # to prevent Wi-Fi credentials accidentally being world-readable.
  chmod 600 ${CONNFILE}
)
```
Nothing in set_wlan() configures /etc/wpa_supplicant/wpa_supplicant.conf. This means that nothing is setting up /etc/wpa_supplicant/wpa_supplicant.conf, which means that the pi cannot connect to wifi.

I do not have the knowledge to verify my hypothesis is correct but regardless in my case /etc/wpa_supplicant/wpa_supplicant.conf was not setup and I have to manually configure it.

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

6.11.0-rc4-7-MANJARO

### Selected OS

Raspberry Pi OS Lite Bookworm 64-bit

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-09-08:
In Raspberry Pi OS Bookworm, we switched from using wpa_supplicant to using NetworkManager instead. See https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

jinxin0, 2024-09-08:
I have no idea then. I got it to work yesterday after editing wpa_supplicant.conf. SSH into the pi, updated it, reboot and it was still working. Then I shut it down with `sudo shutdown now`. Today I try turning it on again to no success.

lurch, 2024-09-08:
This sounds like the sort of thing that you should probably ask for help with on the forums https://forums.raspberrypi.com/

jinxin0, 2024-09-09:
Alright ty.

tdewey-rpi, 2024-09-09:
Closing as 'Won't fix' - we know that WiFi customisation is operational (among other things, I use it nearly every day). In this case, possible user confusion around the migration to NetworkManager was likely the cause of the confusion.

Hope you find the solution in the forums, @jinxin0.

closed 2024-09-09 by tdewey-rpi