raspberrypi/rpi-imager #918: [FEATURE]: options for additional packages and files

somedamnthing, 2024-09-09:
### Is your feature request related to a problem? Please describe.

The problem is, I have a number of Pi's in a lab that I frequently reimage. This means frequently performing first-boot tasks, like update/upgrade, adding packages, and adding arbitrary files.

### Describe the solution you would like to see implemented

I'd like the Pi Imager to provide a field for new packages, as well as the ability to add small text files, in either the General or Services tab.

### Describe alternatives you've considered

Copy/paste: hassle, ok for small `apt` strings, not so much for small files
File transfer: requires setting up some kind of file share, potentially needing more client packages to be installed
Ansible: overwrought for this use case; mixing two system initializers is bad practice

### Additional context

I hope this isn't considered scope creep; it honestly does seem in scope for basic initial setup. This would simply make frequent reimaging an even better 'set and forget' process (apart from hostnames).

### Version

1.8.5 (Default)

lurch, 2024-09-09:
I _suspect_ this probably falls under the same umbrella as #418 , #837 , etc.

somedamnthing, 2024-09-09:
If, [as you say](https://github.com/raspberrypi/rpi-imager/issues/418#issuecomment-2334035599), some kind of first boot solution has been requested multiple times, why has it been rejected multiple times? It's clearly something users of all stripes could find useful. 

As far as the compatibility argument is concerned, I never had any idea/illusion the Imager first boot features would work with any of the other OSs I've tried, so I've just hit 'No'. In fact, everything I've read from RPi representatives on the matter of other OSs is "please consider using the official OS instead".

I don't think anyone expects or particularly wants a fully featured image customizer down to the last py wheel. An open ended, optional first boot script in the style of cloud-init's user-data, where we're allowed to either add polish or make our own dumb mistakes, is the best thing I could ask for.

tdewey-rpi, 2024-09-11:
Feature request rejected.

I've considered this numerous times - and have received no arguments that supplant the safety concern.

Fundamentally, this feature is wide open for abuse, where new users would find a blog that includes a series of instructions that would have them install malware. I consider it unlikely that new users would have the technical competency required to identify malicious instructions, and so the ability of Imager to identify malicious instructions would be paramount.

Unfortunately, I can't see that happening in any useful way.

I'm happy to be proven wrong, and will review a PR - but be advised that a PR will be rejected if it fails to consider user safety.

closed 2024-09-11 by tdewey-rpi