raspberrypi/rpi-imager #927: [FEATURE]: Show and edit OS customization settings with no media

danielhoherd, 2024-09-15:
### Is your feature request related to a problem? Please describe.

As a user, I want the ability to review the configuration settings that are configured for the OS image I have just written. Currently this is not possible unless valid destination media has been inserted into the workstation and selected as a target for writing the OS to.

### Describe the solution you would like to see implemented

I would like to see a button in the UI that lets me view and/or edit the OS customizations that are configured. This may be OS specific, so I could see this being a way to view configurations that were used in the last write, or a way to select an OS image and view the customizations that are configured for that OS. In any case, being able to double-check the configuration settings that were used when writing the prior image is the most important point.

### Describe alternatives you've considered

I could take out my sd card and put it back in and pretend to want to write to it again.

I could find a spare sd card and put it in and pretend to want to write to it.

I could dig through whatever files are on disk and view them, eg `defaults read "${HOME}/Library/Preferences/org.raspberrypi.Imager.plist"`

### Additional context

N/A

### Version

1.9.0 (Default)

tdewey-rpi, 2024-09-18:
Thanks for the suggestion, @danielhoherd 

I removed this functionality in the 1.8 series releases, because there was no reliable and clear way to connect that button to the selected OS - without which connection users may believe an OS was being customised even when that was not possible.

Reintroducing this feature still has the same problem - how to make it clear across the range of cultures and languages that use Raspberry Pi Imager that the customisation will only be applied for a supported OS.

The only way forward I can see is to perform a large-scale UI change - the sort that will merit a major release. As such, I'm neither rejecting nor accepting this feature request, but I will be keeping it in consideration for a future milestone.

tdewey-rpi, 2024-09-18:
> @tdewey-rpi what about being able to see the customizations after selecting the OS, but without having to select the destination media? That seems to be main limiting factor from my perspective as a user.

I could see that possibly working with a grey-enable button, so I'll have a think about it and build a prototype.

headbank, 2025-01-09:
Not quite the same motivation as OP but it strikes me this would dovetail with what I came here to ask about, namely customisation profiles being (ex)portable so installs can be standardised/reproducible. I'll probably post my own ticket about that so as not to scope-creep here, but subscribing as I see a potential overlap of interests.