raspberrypi/rpi-imager #932: [FEATURE]: [DOCS]: 1.9.0: Rebuilding AppImage files

RobertCNelson, 2024-09-18:
### Is your feature request related to a problem? Please describe.

Currently the Directions for Building an AppImage, skip over the AppImage creation tool, along with how it was built:

As a user from debian debuild -us -uc build scripts, what AppImage tool is recommended, and what is run to generate the final Raspberry_Pi_Imager-*.AppImage file?

```
cd rpi-imager
export LD_LIBRARY_PATH=${your_Qt6_install_path}/lib
./${your_platform_appimagecraft_AppImage_name}
```

Now mark the AppImage as executable, and run it:

```
chmod +x ./Raspberry_Pi_Imager-*.AppImage
./Raspberry_Pi_Imager-*.AppImage
```

### Describe the solution you would like to see implemented

Docs: Cleanup AppImage docs

### Describe alternatives you've considered

New AppImage user

### Additional context

Docs

### Version

1.9.0 (Default)

tdewey-rpi, 2024-09-19:
Thanks for the report, @RobertCNelson 

I'll make the docs more explicit - but the answer is in the issue you've raised:

your_platform_**_appimagecraft_**_AppImage_name

Available here: https://github.com/TheAssassin/appimagecraft

It's worth noting, however, that if you want to build for Aarch64 you're better served using my fork, at least until upstream gets to my PRs: https://github.com/tdewey-rpi/appimagecraft/tree/aarch64

tdewey-rpi, 2024-09-19:
Closing as answered. Doc expansion may follow - however the aarch64 outstanding PRs do complicate any flow.

closed 2024-09-19 by tdewey-rpi