raspberrypi/rpi-imager #934: [BUG]: Website still linking to v1.8.5

pelicanmedia, 2024-09-27:
### What happened?

All the OS links on the [website](https://www.raspberrypi.com/software/) are still downloading v1.8.5
[image]

### Version

1.9.0 (Default)

### What host operating system were you using?

macOS

### Host OS Version

NA

### Selected OS

NA

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-09-30:
Thanks for the report, @pelicanmedia 

As v1.9.0 was a major packaging change (including a rotation of our signing key), I wanted to perform a 'soft launch' by releasing it on GitHub for enthusiasts ahead of a more general release.

The initial feedback has so far been overwhelmingly positive, and I've not seen any sign of a showstopper bug for a wider release - however there are a few items that I'd like to address (drawing out of bounds, etc), before requesting a website update - and these will follow in v1.9.1, which is is the preparation stages now.

Holding this open until v1.9.1 is released on the website.

tdewey-rpi, 2024-11-21:
A showstopper issue on v1.9.0 was identified in #942 and I've only recently been able to schedule work on Imager to mitigate that. My hope is to release v1.9.2 across all platforms within the next 2 weeks, assuming we do not identify further showstopper bugs.

core-code, 2024-12-11:
since showstoppers just affect linux, the macOS version on the website could be updated?

lurch, 2024-12-12:
Tom has talked elsewhere about "narrowing the support cone", and having the same version of Raspberry Pi Imager available on Windows, MacOS and Linux helps with that.
(#965 is an example of another showstopper bug)

nafanz, 2025-02-15:
Please tell me when the new version is planned and will it be announced in the blog on the site?

lurch, 2025-02-18:
> Please tell me when the new version is planned

How long is a piece of string? 😉  Tom is also very busy with lots of other projects.

tisba, 2025-03-11:
Is there any release since 1.8.5 available for testing? Happy to help out, if I can.

jseyfert3, 2025-03-12:
So running the 1.8.5 .deb linked on the website on Linux Mint 22.1 I get a "Error: Dependency is not satisfiable: libqt5core5a (> 5.15.1)" error. This sounds similar to #826 and #197, except unlike those cases since I'm running Linux Mint 22.1 which is based on Ubuntu 24.04 it seems like I shouldn't be having this issue.

I don't want to do a new bug report on this though since 1.9.x is supposed to replace 1.8.5, and the 1.9.0 AppImage seems to be working for me. But I did just want to mention it here, since 1.8.5 is still on the website and that's what most users (including me) will find first.

Also, I want to add that while 1.9.0 imager recommends the 64-bit OS, the OS [download page](https://www.raspberrypi.com/software/operating-systems/) still says that the 32-bit version is recommended. I'd suggest that some wording be added that the 64-bit is recommended unless you have an incompatible Pi. It looks like this was planned, but perhaps got missed, as [this post](https://forums.raspberrypi.com/viewtopic.php?t=357532#p2143342) from Oct 2023 indicates.

lurch, 2025-03-13:
> Also, I want to add that while 1.9.0 imager recommends the 64-bit OS, the OS [download page](https://www.raspberrypi.com/software/operating-systems/) still says that the 32-bit version is [...]

We want the initial set-up process to be as low-friction for new users as possible. The OS download page obviously has no idea which Pi a user is using, so it recommends the 32-bit OS because that will at least boot on all models of Raspberry Pi. If somebody new to the whole Raspberry Pi ecosystem buys e.g. a Raspberry Pi Zero, and downloads the 64-bit OS image (which might take a while for somebody with a slower internet connection), writes it to a microSD card, and finds that it doesn't boot, they might think that their new Pi is broken, rather than realising that they simply downloaded the wrong OS image.

OTOH, Raspberry Pi Imager **does** know which Pi a user is using (via the "Choose Device" filter), and so there we're able to recommend the 32-bit or 64-bit OS image as appropriate.

Of course this is all IMHO, and we may or may not change this stance in future.

nafanz, 2025-06-05:
@tdewey-rpi maybe it's worth writing a blog post about this? there hasn't been any news about this utility for a long time.