raspberrypi/rpi-imager #938: [BUG]: Raspberry Pi Imager phones home with surveillance without my consent like spyware

sneak, 2024-09-30:
### What happened?

The imager covertly exfiltrates my activity data to the hostname `rpi-imager-stats.raspberrypi.com` without consent when I flash an image.  

This happens silently, so not only does it happen against my will and without my consent, it doesn't even disclose that it's doing so, just like spyware.

### Version

1.8.5

### What host operating system were you using?

macOS

### Host OS Version

15.0

### Selected OS

HAOS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-09-30:
https://github.com/raspberrypi/rpi-imager?tab=readme-ov-file#telemetry

sneak, 2024-10-01:
Yes, I understand that the nonconsensual stealth surveillance is intentionally included in the app.  That's wrong.