raspberrypi/rpi-imager #941: [BUG]: Image customization assumes cloud-init on local images

steev, 2024-10-02:
### What happened?

Heya!  

I'm working on getting the [customized options support in Kali Linux](https://gitlab.com/kalilinux/build-scripts/kali-arm/-/issues/317) and what I've found is that if I write a release image, because we have `init_format` set to `systemd` in our json file, it does create the script.

However, if I attempt to use a locally created image, it seems like it defaults to thinking that the `init_format` is set to `cloud_init`

I did see https://github.com/raspberrypi/rpi-imager/pull/832 where you were considering just disabling customization, but it seems the conversation was taken offline, and I probably missed the updated pull request where this may now be worked around.

I am using 1.8.5 still because I get an undefined symbol with the 1.9.0 aarch64 AppImage.

### Version

1.8.5

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Kali Linux 2023.4

### Selected OS

Custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-10-03:
Thanks for the report, @steev 

I'd be deeply interested in your AppImage launch failure - could you please raise a second issue for that? At first guess, it's going to be a glibc mismatch?

tdewey-rpi, 2024-10-03:
Expanding on this further - if you're looking to add customisation support, I'd strongly encourage you to go down the cloud-init path. I'm currently pushing the Raspberry Pi OS toward cloud-init, with a view to marking the firstrun.sh-style customisation as deprecated in an upcoming Imager.

That said, using the 'custom image' function I'd absolutely not expect customisation to be applied - the window shouldn't be shown, nor should the choice to use customisation. If you're seeing any sign of customisation in this case, that's a concrete bug and needs to be squashed.

steev, 2024-10-05:
I definitely see it with a custom image I spin up on my local machine and 

Choose Device-> RaspberryPi 5   
Choose OS -> Use custom -> select from the location on my computer  
Choose Storage -> choose the SDcard  
Next

I get prompted "Use OS Customization?"

tdewey-rpi, 2024-10-07:
> I definitely see it with a custom image I spin up on my local machine and
> 
> Choose Device-> RaspberryPi 5 Choose OS -> Use custom -> select from the location on my computer Choose Storage -> choose the SDcard Next
> 
> I get prompted "Use OS Customization?"

Ooof! Yep, that's not what I expected at all. I'm going to take two actions as part of this issue:

1) Identify the cloud-init default mode, because that doesn't feel safe for later development
2) Reintroduce or fix up restrictions on customising the 'custom image' flow - the local repo flow will retain that capability.

steev, 2024-10-07:
> Ooof! Yep, that's not what I expected at all. I'm going to take two actions as part of this issue:
>
> 1. Identify the cloud-init default mode, because that doesn't feel safe for later development

Will we have to add network_renderer or something along these lines for our json?  I don't mind testing with networkd, but me testing something and end users using something is a whole different ball game ;)

> 2. Reintroduce or fix up restrictions on customising the 'custom image' flow - the local repo flow will retain that capability.

That makes sense.

As I noted in #942 once I was able to test 1.9.0 by using the Xorg backend for GNOME, I see the same on 1.9.0, it prompts me for customization options with a custom image.

cmd0s, 2024-10-20:
I have another use case where I intentionally use both "custom image" and "Use OS Customization" My image is a slightly modified version of Ubuntu, and the "Use OS Customization" option is still useful and even necessary for me.

I’m a developer on the Web3 Pi project, where, in simple terms, we provide a specialized image for Raspberry Pi. Once this image is flashed onto a memory card and the Raspberry Pi is booted, the installer will automatically configure and launch a Full Ethereum Node.

From my perspective, it would be ideal if you could avoid removing the "Use OS Customization" functionality for custom images.

You can find more information on our website: [https://www.web3pi.io/welcome-box](https://www.web3pi.io/welcome-box)

tdewey-rpi, 2024-10-22:
> I have another use case where I intentionally use both "custom image" and "Use OS Customization" My image is a slightly modified version of Ubuntu, and the "Use OS Customization" option is still useful and even necessary for me.
> 
> I’m a developer on the Web3 Pi project, where, in simple terms, we provide a specialized image for Raspberry Pi. Once this image is flashed onto a memory card and the Raspberry Pi is booted, the installer will automatically configure and launch a Full Ethereum Node.
> <snip>
> You can find more information on our website: https://www.web3pi.io/welcome-box

Have you considered listing your image in the Raspberry Pi Imager OS list? That would simplify your deployments _and_ retain the option for OS customisation in a way that doesn't expose issues of guessing incorrectly.

cmd0s, 2024-10-22:
> > I have another use case where I intentionally use both "custom image" and "Use OS Customization" My image is a slightly modified version of Ubuntu, and the "Use OS Customization" option is still useful and even necessary for me.
> > I’m a developer on the Web3 Pi project, where, in simple terms, we provide a specialized image for Raspberry Pi. Once this image is flashed onto a memory card and the Raspberry Pi is booted, the installer will automatically configure and launch a Full Ethereum Node.
> > 
> > You can find more information on our website: https://www.web3pi.io/welcome-box
> 
> Have you considered listing your image in the Raspberry Pi Imager OS list? That would simplify your deployments _and_ retain the option for OS customisation in a way that doesn't expose issues of guessing incorrectly.

Thanks for your message! It would indeed be fantastic if the Web3 Pi image could be selected directly from the Raspberry Pi Imager OS list. Is there any documentation or instructions on how to submit/add an image to the Raspberry Pi Imager application?

tdewey-rpi, 2024-10-22:
> Thanks for your message! It would indeed be fantastic if the Web3 Pi image could be selected directly from the Raspberry Pi Imager OS list. Is there any documentation or instructions on how to submit/add an image to the Raspberry Pi Imager application?

Best plan is to fill out this form, which includes an example of the sort of file you'd need to generate & host. Many projects choose to host the subitems JSON file on GitHub - and if you do, please remember to use the raw.github URL!

https://forms.gle/X7HpED5fYcjW1k8c7

dp1140a, 2025-08-05:
I love the idea of being able to use cloud-init.  I install Ubuntu Server 22 or 24 on CM4 modules.  Sometimes the customization works and sometimes not so much.  Being able to just create a cloud-init config and point imager to it would be awesome.  Anyone done this?  If so please point it out.  Would be a huge help.

closed 2025-12-12 by tdewey-rpi