raspberrypi/rpi-imager #942: [BUG]: symbol lookup error with 1.9.0 and mesa 24.2.2-1 ( Kali Linux host, Debian testing equivalent)

steev, 2024-10-04:
### What happened?

When attempting to run the aarch64 1.9.0 AppImage, I am getting an unknown symbol

```
steev@finn:~/Downloads$ ./Raspberry_Pi_Imager-1.9.0-aarch64.AppImage
/tmp/.mount_RaspbeVM7Y30/AppRun.wrapped: symbol lookup error: /lib/aarch64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
```

I am guessing this went away in newer mesa, the version of libegl-mesa0 is 24.2.2-1 from Debian testing.

### Version

1.9.0 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Kali Linux 

### Selected OS

Kali Linux

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-10-07:
Thanks for the report, @steev 

This looks like a fairly major problem - the issue doesn't go away with a newer mesa - it appears to be _introduced_ by it.

Raspberry Pi OS is shipping today with 23.2.1 (+rpi changes). It appears that the libwayland-client that is embedded in the AppImage is missing the symbol you've referenced - causing the failure.

I note that the PCSX2 project encountered a similar problem: https://github.com/PCSX2/pcsx2/issues/11407

Their solution was to deliberately exclude the libwayland-* libraries, and I'm tempted to do the same after some testing.

Is this Kali image now published in Imager?

steev, 2024-10-07:
> Thanks for the report, @steev
> 
> This looks like a fairly major problem - the issue doesn't go away with a newer mesa - it appears to be _introduced_ by it.
> 
> Raspberry Pi OS is shipping today with 23.2.1 (+rpi changes). It appears that the libwayland-client that is embedded in the AppImage is missing the symbol you've referenced - causing the failure.
> 
> I note that the PCSX2 project encountered a similar problem: [PCSX2/pcsx2#11407](https://github.com/PCSX2/pcsx2/issues/11407)
> 
> Their solution was to deliberately exclude the libwayland-* libraries, and I'm tempted to do the same after some testing.
> 
Ah, I didn't even think to try with Xorg as the GNOME backend.  With Xorg as the GNOME backend, indeed, the error does not show up.  I just assumed it was mesa because of the libEGL complaint.

> Is this Kali image now published in Imager?

Unfortunately no, not yet.  I've never used cloud-init before so I'm still learning it.  Once I have something that people can test, it will probably be part of our weekly images, and I'll tell users how to create a json file with the bits to use.

closed 2024-12-06 by tdewey-rpi

nafanz, 2025-02-15:
I use:
* OS: Ubuntu 24.10 x86_64
* Kernel: Linux 6.11.0-14-generic
* DE: GNOME
* WM: Mutter (Wayland)

I have a similar error when starting:
```
./Raspberry_Pi_Imager-1.9.0-x86_64.AppImage 
/tmp/.mount_Raspbey2Kwwy/AppRun.wrapped: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
```

Am I right in understanding that starting with version 1.9.0 and above this will no longer work with Wayland, meaning there should be no fixes?

lurch, 2025-02-18:
> Am I right in understanding that starting with version 1.9.0 and above this will no longer work with Wayland, meaning there should be no fixes?

According to #975 the answer appears to be "yes".

tdewey-rpi, 2025-02-18:
Confirming: That is correct. 1.9.3 will remove the Wayland back-end from Imager, and default to using XCB - which on Wayland platforms will result in Imager -> XWayland -> Wayland.

PR #975 (2024-12-06): AppImage: Disable Wayland support
The Wayland composition flags in the Qt AppImage support are not safe to use - they include a system-owned copy of libwayland-client that is cross-called by the system Mesa - outside of the AppImage. Including the system-owned copy from the build host creates an ABI breakage that prevents execution.

This is an unacceptable situation, and until a more precise fix can be provided, disable Wayland client support in Imager.

Fixes #942