raspberrypi/rpi-imager #943: [BUG]: macOS Sequoia auth issue

JonathanGSL, 2024-10-08:
### What happened?

After selecting “Yes, I know the drive will be erased” (or whatever the message is) the auth prompt is displayed for the user to use TouchID but the app meanwhile says auth failed, and setting disk access is needed.

### Version

1.8.5

### What host operating system were you using?

macOS

### Host OS Version

15.0.1 (24A348)

### Selected OS

Bookwork Lite 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-10-08:
Thanks for the report, @JonathanGSL 

Can you confirm if you see the same behaviour with the soft-launched 1.9.0?

https://github.com/raspberrypi/rpi-imager/releases/tag/v1.9.0

tdewey-rpi, 2024-11-01:
@JonathanGSL Did you find you had the same problem with v1.9.0?

JonathanGSL, 2024-11-05:
Hi Tom.

Sorry for the delay. No, I couldn’t reproduce it with 1.9 .

Kind regards, keep safe…

Jonathan

Jonathan C. L. Groves
Director
Groves Solutions Ltd
Groves-Solutions.co.uk

0(+44)7970 468 354

[image001.png]

The content (and any attachments) of this email is (are) intended for the recipient only - please do not distribute without permission.
If you are not the intended recipient, please delete this message.
©Groves Solutions Ltd 2024 – All Rights Reserved.

tdewey-rpi, 2024-11-19:
Closing as 'Fixed' - handled in the 1.9.x series releases.

closed 2024-11-19 by tdewey-rpi