raspberrypi/rpi-imager #945: [BUG]: Filename shortening if file is created by imager

paulober, 2024-10-09:
### What happened?

Cloud-init network-config can't be created by imager only edited if already present in the imager because it would shorten the filename which results in an invalid filename for the cloud-init datasource.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

24H2

### Selected OS

Raspberry Pi OS Bookworm Lite (64-bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
https://github.com/raspberrypi/rpi-imager/blob/879b89cae1e35d5d6dd6b26aa269a0f3e260a5c3/src/devicewrapperfatpartition.cpp#L402-L483
```

lurch, 2024-10-10:
What filename is it supposed to have? What does it get shortened to?

maxnet, 2024-10-10:
>In practice, we see Imager producing short filenames essentially all the time.

Pretty sure it did generate new files like "firstrun.sh" (lowercase) correctly in the past.
If storing long file name entries would not work at all, I would expect it to show up as FIRSTRUN.SH instead, as the 8.3 short name is stored in uppercase.

closed 2025-01-08 by tdewey-rpi