raspberrypi/rpi-imager #946: [FEATURE]: Pass config via CLI options

pajasry, 2024-10-13:
### Is your feature request related to a problem? Please describe.

I would like to automatize the deployment of pi images for my projects. It would be usefull to be able configure WiFi and other setting using script instead of UI.

### Describe the solution you would like to see implemented

Pass config.txt to image writer in CLI mode

### Describe alternatives you've considered

Making custom first-run script, but it is not ideal solution.

### Additional context

The config is parsed only in GUI mode, in CLI mode is always ommited. See https://github.com/raspberrypi/rpi-imager/blob/879b89cae1e35d5d6dd6b26aa269a0f3e260a5c3/src/cli.cpp#L211 and https://github.com/raspberrypi/rpi-imager/blob/879b89cae1e35d5d6dd6b26aa269a0f3e260a5c3/src/cli.cpp#L189

### Version

1.9.2 (Default)

tdewey-rpi, 2024-10-18:
Thanks for the report, @pajasry

CLI already supports using the `cloud-init` configuration scheme, as your included line highlights. This mechanism is currently supported on Ubuntu Server only (I believe), but we are looking to change Raspberry Pi OS to use `cloud-init`, which will also enable this use-case.

As the `firstrun` style customisation is planned for deprecation, I'm inclined to reject this feature request on two grounds:

1. `firstrun` is on the way out - and unless there are specific security or ergonomic issues affecting a wide range of users I'm not inclined to modify it
2. `cloud-init` configuration is already available

That said, I could be convinced otherwise - but only if you can demonstrate a solid reason to put time into the `firstrun` mechanism.

tdewey-rpi, 2024-10-29:
Resolving as 'Rejected'. Rationale is included in the above post, and I have not seen a compelling reason to change course.

closed 2024-10-29 by tdewey-rpi