raspberrypi/rpi-imager #951: [BUG]: The text wrapping is not working correctly

cmd0s, 2024-10-18:
### What happened?

As shown in the attached screenshot, when there is a long file name for an image, the text in the "Write Successful" label does not fit within the window and extends beyond its boundaries. The text should be wrapped onto a new line. It's a minor detail, but it should certainly be corrected.

[image]

### Version

1.9.0

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Ubuntu 24.04.1

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-10-18:
Thanks for the report, @cmd0s 

This is a known issue with v1.9.0, and something I'm expecting to resolve (or at least mitigate) on v1.9.2.

Curiously, it only seems to happen on the Windows platform - macOS and Linux platforms do not encounter this problem.

tdewey-rpi, 2024-11-27:
Confirmed this is resolved in $HEAD, will close after next release.