raspberrypi/rpi-imager #954: [FEATURE]: Install and register rpi-connect

GeekyTim, 2024-10-23:
### Is your feature request related to a problem? Please describe.

Not problem - just a useful enhancement.

### Describe the solution you would like to see implemented

I would like rpi-connect to be an option for installation when writing a new RPi OS image, with the server being registered with the rpi-connect server.

### Describe alternatives you've considered

I install rpi-connect on most of my servers.

### Additional context

This feature would save time when building new servers.

### Version

1.9.0

ghollingworth, 2024-10-24:
Yes, this is something we've been intending to be able to do at sometime in the future, but it requires some extra support from Pi Connect.  Currently, the token you create when you connect to a new device is attached to the device's serial number, the idea is to stop someone from supplying you an SD card pre-provisioned so they have access to your hardware!

This would require something different, like a passkey or a special time limited un-serialized token (i.e. you have a token that can only be used once to generate a new real token but it must be used within some time period)

But it is on the roadmap...

I'll close this (mostly just so we don't end up with a gajillion issues!), but it is something we're going to enable...

Gordon

closed 2024-10-24 by ghollingworth

GeekyTim, 2024-10-24:
Excellent!