raspberrypi/rpi-imager #956: [BUG]: firstrun.sh: line72: unexpected eof while looking for matching `''

jujax, 2024-11-03:
### What happened?

Hello,
I flashed raspberry os lite (I tried 64 and 32bits) on a raspberry pi 3 with imager v1.8.5 and with personnal settings like ssh key, wifi credentials etc ...
My raspberry stop at boot after lanching firstrun.sh, this is the error I see on my screen:

> /boot/firmware/firstrun.sh: line72: unexpected eof while looking for matching `''

after that line kernel-command-line.service failed to start and the os is powered off with shutdown.target

I also tried to flash raspbian with my settings and it's working

### Version

1.8.5

### What host operating system were you using?

macOS

### Host OS Version

Sequoia, 15.1

### Selected OS

Raspberry Pi OS Lite Bookworm 64 bits

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
/boot/firmware/firstrun.sh: line72: unexpected eof while looking for matching `''
```

jujax, 2024-11-03:
This is firstrun.sh file on my sd card:

#!/bin/bash

set +e

FIRSTUSER=`getent passwd 1000 | cut -d: -f1`
FIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh -k 'dD"'"'"'Éssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAACAQDBGDLcRNcAnSDVYGejdFaqwf3lqCPLa864l4//FPBKVEhhGcB6pa0kD2s5ON4VxXoCBwsnRyjvafunkkXC1RZOsnfWt0At9d+2fKypoClQflBKvFRgHAcVfIKGsaLHY7L/JTBzKRhF4FDP6cOB56FxstqL+ubbq9u3DumTXtU2Yqrv4DDQudDNmEKCgAfU1Gzq7zH7+OGLBMvGgMpGMOxDh7/8LFrEhlVlnizVK2/dXnI/Y7L4HSyyutfX+5OZMbwK7m14nc3x8bLngsz6+9kGoeqUBAH8ZMbH08JM/qED2gOwzS+bXXDxGKVWQ/GIQx5Ur6UxsfqX6lNZopPlLoeEiUJOGBCWtnHkKC7VVdHy7ugAjKrLYtFH9pFFbEx4dtNcrCF20H6cF3rjgLaepme5oCNmtEw/OGoU0SKMXqPJRnneIRFVdia8uNqzjjCkI7O9eHz8emhjY+TscKF+rM38egxzjqRPcLbqw2hlOVA52MBlkzW08zD63c6t0o0JmY75bGc2aO8XO3HT4UO5PUEOw/mtT6HSX9RpmlyZIPQM/fBGGfblPXEMIrRnNQk5H4vxts8yNZmmclsUm+JFm1AZAF2dL78KYBYKI1R3PyLBsL7EOdrxiugS4dnyZrZqabEs1YczMz3KjyG+9I4kzECiEUPPVucoNW7eo5SGcRdw9Q== pelletier.julien@pm.me'
else
   install -o "$FIRSTUSER" -m 700 -d "$FIRSTUSERHOME/.ssh"
   install -o "$FIRSTUSER" -m 600 <(printf "dD"'Éssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAACAQDBGDLcRNcAnSDVYGejdFaqwf3lqCPLa864l4//FPBKVEhhGcB6pa0kD2s5ON4VxXoCBwsnRyjvafunkkXC1RZOsnfWt0At9d+2fKypoClQflBKvFRgHAcVfIKGsaLHY7L/JTBzKRhF4FDP6cOB56FxstqL+ubbq9u3DumTXtU2Yqrv4DDQudDNmEKCgAfU1Gzq7zH7+OGLBMvGgMpGMOxDh7/8LFrEhlVlnizVK2/dXnI/Y7L4HSyyutfX+5OZMbwK7m14nc3x8bLngsz6+9kGoeqUBAH8ZMbH08JM/qED2gOwzS+bXXDxGKVWQ/GIQx5Ur6UxsfqX6lNZopPlLoeEiUJOGBCWtnHkKC7VVdHy7ugAjKrLYtFH9pFFbEx4dtNcrCF20H6cF3rjgLaepme5oCNmtEw/OGoU0SKMXqPJRnneIRFVdia8uNqzjjCkI7O9eHz8emhjY+TscKF+rM38egxzjqRPcLbqw2hlOVA52MBlkzW08zD63c6t0o0JmY75bGc2aO8XO3HT4UO5PUEOw/mtT6HSX9RpmlyZIPQM/fBGGfblPXEMIrRnNQk5H4vxts8yNZmmclsUm+JFm1AZAF2dL78KYBYKI1R3PyLBsL7EOdrxiugS4dnyZrZqabEs1YczMz3KjyG+9I4kzECiEUPPVucoNW7eo5SGcRdw9Q== pelletier.julien@pm.me") "$FIRSTUSERHOME/.ssh/authorized_keys"
   echo 'PasswordAuthentication no' >>/etc/ssh/sshd_config
   systemctl enable ssh
fi
if [ -f /usr/lib/userconf-pi/userconf ]; then
   /usr/lib/userconf-pi/userconf 'julien' '****'
else
   echo "$FIRSTUSER:"'****' | chpasswd -e
   if [ "$FIRSTUSER" != "julien" ]; then
      usermod -l "julien" "$FIRSTUSER"
      usermod -m -d "/home/julien" "julien"
      groupmod -n "julien" "$FIRSTUSER"
      if grep -q "^autologin-user=" /etc/lightdm/lightdm.conf ; then
         sed /etc/lightdm/lightdm.conf -i -e "s/^autologin-user=.*/autologin-user=julien/"
      fi
      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then
         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e "s/$FIRSTUSER/julien/"
      fi
      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then
         sed -i "s/^$FIRSTUSER /julien /" /etc/sudoers.d/010_pi-nopasswd
      fi
   fi
fi
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan '****' '****' 'FR'
else
cat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'
country=FR
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

update_config=1
network={
	ssid="****"
	psk=****
}

WPAEOF
   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
   rfkill unblock wifi
   for filename in /var/lib/systemd/rfkill/*:wlan ; do
       echo 0 > $filename
   done
fi
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_keymap 'fr'
   /usr/lib/raspberrypi-sys-mods/imager_custom set_timezone 'Europe/Paris'
else
   rm -f /etc/localtime
   echo "Europe/Paris" >/etc/timezone
   dpkg-reconfigure -f noninteractive tzdata
cat >/etc/default/keyboard <<'KBEOF'
XKBMODEL="pc105"
XKBLAYOUT="fr"
XKBVARIANT=""
XKBOPTIONS=""

KBEOF
   dpkg-reconfigure -f noninteractive keyboard-configuration
fi
rm -f /boot/firstrun.sh
sed -i 's| systemd.run.*||g' /boot/cmdline.txt
exit 0

jujax, 2024-11-03:
my ssh public key was corrupted ...

tdewey-rpi, 2024-11-05:
Thanks for the report, @jujax.

Can you confirm if you can reproduce this issue?

jujax, 2024-11-06:
Yes I put `qffsefesf'""'"test` in ssh input and boot is falling again

lurch, 2024-11-06:
What a strange password to choose... :wink:  :joy:

jujax, 2024-11-06:
It's in ssh public key and not password sorry for the mistake

lurch, 2024-11-06:
Those characters aren't valid in SSH public keys anyway, so perhaps the correct fix here is to add better input-validation in the OS Customisation dialog, rather than trying to fix `firstrun.sh` to cope with escaping extra characters?