raspberrypi/rpi-imager #957: [FEATURE]: Add option to backup image file from an SD Card

touhidurrr, 2024-11-03:
### Is your feature request related to a problem? Please describe.

Usually after booting to Raspberry Pi for the first time, I like to do a lot of setup specific to myself. Setting up Node.js from node source, installing Bun etc. However, often I would need to redo all the steps again for random issues like SD Card getting corrupted. Or poor management of the SD Cards I own etc. I would love if I could somehow make a backup of my Pi as an image which I could then write to other SD Cards to retain my setup. This way I will not lose my setup if I ever lose my SD Card or get it corrupted again.

### Describe the solution you would like to see implemented

Add an option to write an `.img.xz` file from an SD Card.

### Describe alternatives you've considered

Read tutorials to learn how to use dd to make an image. Then run the commands in Pi. Then move that image from Pi to my PC for backup. (Too Many Steps!!!)

### Additional context

None

### Version

1.9.0

lurch, 2024-11-04:
This has already been requested (and rejected) multiple times https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aissue+backup+is%3Aclosed

tdewey-rpi, 2024-11-05:
Closing, rationale per @lurch's comment.

closed 2024-11-05 by tdewey-rpi