raspberrypi/rpi-imager #958: [BUG]: Latest Version of Raspbian is not correctly written with WiFi credentials

deathrjj, 2024-11-05:
### What happened?

Imaging using the version of Raspbian released 2024-10-22 WiFi credentials are no longer correctly applied. 

This makes headless setup much more difficult.

Have tested with a Raspberry Pi 3b and multiple Raspberry Pi 4, issue occurs on all.

### Version

1.9.0

### What host operating system were you using?

macOS

### Host OS Version

Sequoia 15.1

### Selected OS

Raspberry Pi Os Bookworm 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-11-05:
I tested this myself with the latest image just last week (on a Zero 2 W, although that shouldn't matter), and it worked fine. Are you 100% sure that you didn't make a typo in the SSID or password?

deathrjj, 2024-11-05:
I feel so stupid the SSID was missing a capital letter.

closed 2024-11-05 by deathrjj

lurch, 2024-11-05:
No worries, silly mistakes happen to all of us at some point or another :grinning: