raspberrypi/rpi-imager #961: [BUG]: Imager: Unknown option 'version'.

labbedaine, 2024-11-16:
### What happened?

Hello!

I have a Raspberry Pi OS Lite installed on an SD card and I am using it to install a Raspberry Pi OS Lite on an NVME disk.  However when I run commands I get an error.

`sudo rpi-imager --cli --version` --> Imager: Unknown option 'version'.

Version (putting it here since 1.9.1 is not an option on your dropdown menu): **rpi-imager is already the newest version (1.9.1).**

Thank you!

### Version

1.9.0

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Bookworm

### Selected OS

raspios_lite_arm64

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

NVMe SSD via Third-party PCIe attachment

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-11-16:
Looks like `rpi-imager --cli --version` doesn't work, but `rpi-imager --version` works just fine?
```console
$ rpi-imager --help
rpi-imager [--debug] [--version] [--repo <repository URL>] [--qm <custom qm translation file>] [--disable-telemetry] [<image file to write>]
-OR- rpi-imager --cli [--disable-verify] [--sha256 <expected hash>] [--debug] [--quiet] <image file to write> <destination drive device>
$ rpi-imager --version
rpi-imager version 1.9.1
Repository: https://downloads.raspberrypi.org/os_list_imagingutility_v4.json
$ rpi-imager --cli --version
Imager: Unknown option 'version'.
```

lurch, 2024-11-16:
```
myuser@rpi-sbx:~ $ sudo rpi-imager --version
qt.qpa.xcb: could not connect to display 
```
Interesting... @tdewey-rpi Could we tweak `rpi-imager` so that it doesn't try opening a display if all we're doing is printing out the version-string? Or is that all deeply interwoven in the Qt framework and therefore impossible to disable?

> I prefer using CLI...

If you really want to use a CLI, I guess you can always use the good old `dd` command, but make sure you're _very_ careful.

> I can use Imager UI to setup my NVME drive.

I guess that means you're probably using a Pi 5, in which case you could just use the network-boot-mode of RPi Imager to write an OS to your NVMe drive? See https://www.raspberrypi.com/documentation/computers/getting-started.html#install-over-the-network

tdewey-rpi, 2024-11-18:
Network install is absolutely the way to go, if you can maintain an internet connection, for NVME hardware - among other things, it makes the process of configuring the boot order somewhat straightforward.

tdewey-rpi, 2024-11-18:
Concerning the Qt QPA dependencies - I suspect this is going to be a fairly involved change - the Qt QPA dependency isn't an explicit thing we set. Instead, it's an implicit request from the Qt framework - which we also use in the `cli` mode.

I suspect this won't be resolved until a larger rework of the Imager is completed - which is indeed planned, but on a long timeframe while I complete other projects.

lurch, 2024-11-18:
I guess it might make sense to close this issue then, as it sounds like Raspberry Pi Imager isn't supported on Raspberry Pi OS Lite?

tdewey-rpi, 2024-11-21:
Unfortunately, @lurch is correct. I'm going to have to close this as 'Won't fix', because fundamentally, rpi-imager is a graphical tool for creating OS media for Raspberry Pi devices.

That's not saying that in the future, there won't be better support for TUI-style environments - simply that it's not something offered today, and so this isn't really a Bug.

andreibsk, 2025-12-23:
As a workaround for running the `--cli` option on the Lite version of the OS, the missing runtime dependencies can be installed manually. The example below has worked for me as of today:

```
sudo apt install rpi-imager libegl1 libfontconfig1 libglx0 libopengl0 -y
sudo rpi-imager --cli ./2025-12-04-raspios-trixie-arm64-lite.img.xz /dev/nvme0n1
```

tdewey-rpi, 2025-12-23:
> As a workaround for running the `--cli` option on the Lite version of the OS, the missing runtime dependencies can be installed manually. The example below has worked for me as of today:
>
> ```
> sudo [...]

Why not just use the cli package?