raspberrypi/rpi-imager #966: [FEATURE]: OOption to Display Password Entered in the OS Customisation Panel for "Set username and password"

butchkemper, 2024-11-25:
### Is your feature request related to a problem? Please describe.

When entering the username/password, the password is displayed as dots and no option is available to display the typed characters.  When the password is not displayed, it is not possible to see if the desired password was correctly entered.

### Describe the solution you would like to see implemented

The Set username and password section of OS Customisation window should have a "Show password" option that causes the entered password to be displayed. The "Configure wireless LAN" has a "Show password" option to display the Wireless LAN password.

### Describe alternatives you've considered

No alternative is available display the characters entered for the username password.

### Additional context

None

### Version

1.8.5

butchkemper, 2024-11-25:
Correct type

butchkemper, 2024-12-01:
> I believe 1.8.5, and certainly 1.9.0 show the characters of your password as you enter them.
> Showing a stored password isn't a viable mechanism, as we perform a one-way hash of the password upon settings save - so that button could only ever work on the occasion you're entering the password - and would only offer confusion in the following openings of the dialog.
> I generally steer against having dynamic UI that isn't consistent across executions of the program - implied state is a poor UX at best, and creates additional documentation load.
> Are you saying you don't see the characters one at a time as you enter?

Tom,

Thank you for your response and I apologize for my delay in responding.

I can verify that Imager 1.8.5 shows individual password characters as they are entered for a very short time and then the characters become dots.

I checked how Imager worked with the Wireless LAN configuration and the Show password option.  With the option checked, the password characters were displayed when entered and remained displayed until the option was deselected which replaced the characters with dots.  If the option was again selected, the proper characters were displayed.

The LAN password is not encrypted until YES on the Use OS customisation panel is clicked.

I recommend a Show password option for the username password display the entered characters until the password is encrypted.  Encrypted passwords would show as dots.  

The number of dots displayed changes when the password is encrypted and 
that happens when the YES on the Use OS customisation panel is clicked.
 
Imager must know when a password is encrypted to prevent further encryption of an already encrypted password. This encryption status could be used so an encrypted password is displayed as dots when the show password option is selected.

Hopefully, my explanations are clear.

Butch 

--==PART_BOUNDARY_0_1560776429_0_2214509949==
Content-Transfer-Encoding: 8bit
Content-Type: text/html; charset="utf-8"

<HTML XMLNS:emoji><HEAD><META http-equiv="Content-Type" content="text/html; charset=utf-8" />
</HEAD><BODY STYLE="font-family:'Courier New'; font-size:13pt; ">
<DIV> </DIV>
<DIV>Tom,</DIV>
<DIV> </DIV>
<DIV>Thank you for your response and I apologize for my delay in responding.</DIV>
<DIV> </DIV>
<DIV>I can verify that Imager 1.8.5 shows individual password characters as they are entered for <STRONG>a very short time</STRONG> and then the characters become dots.</DIV>
<DIV> </DIV>
<DIV>I checked how Imager worked with the Wireless LAN configuration and the Show password option.  With the option checked, the password characters were displayed when entered and remained displayed until the option was deselected which replaced the characters with dots.  If the option was again selected, the proper characters were displayed.</DIV>
<DIV> </DIV>
<DIV>The LAN password is not encrypted until YES on the Use OS customisation panel is clicked.</DIV>
<DIV> </DIV>
<DIV>I recommend a Show password option for the username password display the entered characters until the password is encrypted.  Encrypted passwords would show as dots.  </DIV>
<DIV> </DIV>
<DIV>The number of dots displayed changes when the password is encrypted and </DIV>
<DIV>that happens when the YES on the Use OS customisation panel is clicked.</DIV>
<DIV> </DIV>
<DIV>Imager must know when a password is encrypted to prevent further encryption of an already encrypted password. This encryption status could be used so an encrypted password is displayed as dots when the show password option is selected.</DIV>
<DIV> </DIV>
<DIV>Hopefully, my explanations are clear.</DIV>
<DIV> </DIV>
<DIV>Butch </DIV>
<DIV> </DIV>
<DIV> </DIV>
<DIV> </DIV>
<DIV>  </DIV>
<DIV>At 05:28 AM 11/27/2024 -0800, Tom Dewey wrote:</DIV>
<DIV>
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px"> </BLOCKQUOTE></DIV>
<DIV>
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">I believe 1.8.5, and certainly 1.9.0 show the characters of your password as you enter them.</BLOCKQUOTE></DIV>
<DIV>
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">Showing a stored password isn't a viable mechanism, as we perform a one-way hash of the password upon settings save - so that button could only ever work on the occasion you're entering the password - and would only offer confusion in the following openings of the dialog.</BLOCKQUOTE></DIV>
<DIV>
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">I generally steer against having dynamic UI that isn't consistent across executions of the program - implied state is a poor UX at best, and creates additional documentation load.</BLOCKQUOTE></DIV>
<DIV>
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">Are you saying you don't see the characters one at a time as you enter?</BLOCKQUOTE></DIV>
<DIV style="FONT-SIZE: small; COLOR: #666; -webkit-text-size-adjust: none">
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">—</BLOCKQUOTE></DIV>
<DIV style="FONT-SIZE: small">
<BLOCKQUOTE style="BORDER-RIGHT-WIDTH: 0px; BORDER-BOTTOM-WIDTH: 0px; PADDING-BOTTOM: 0px; PADDING-TOP: 0px; PADDING-LEFT: 4px; BORDER-LEFT: rgb(137,137,137) 3px solid; MARGIN: 0px; DISPLAY: block; PADDING-RIGHT: 0px; BORDER-TOP-WIDTH: 0px">
<DIV style="FONT-SIZE: small">
<DIV> </DIV></BODY></HTML>

--==PART_BOUNDARY_0_1560776429_0_2214509949==--

tdewey-rpi, 2025-12-12:
Implemented in 2.0.2.

closed 2025-12-12 by tdewey-rpi