raspberrypi/rpi-imager #972: [BUG]: Warning message should be wrapped in PL translation

nonflammable, 2024-12-03:
### What happened?

[image]

### Version

1.9.0

### What host operating system were you using?

Windows

### Host OS Version

21H2

### Selected OS

any

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-12-06:
Thanks for the report, @nonflammable 

The screenshot is particularly helpful, as it helps to more easily identify the UI location that's missing bounds. Expect this to be resolved in v1.9.2.

nonflammable, 2024-12-06:
[image]
in the EN version it is ok, but in the PL version it does not fit

tdewey-rpi, 2024-12-10:
I believe this will be fixed with v1.9.2, as part of a generic fix to the message box object for text wrapping.

tdewey-rpi, 2025-01-15:
Good catch, @lurch.

This is indeed the same issue - even with the same components. Closing this one in favour of resolving #951