raspberrypi/rpi-imager #973: [BUG]: macOS v1.85 and v1.9 "No storage devices found"

davidgs, 2024-12-04:
### What happened?

When attempting to select a storage medium, none are found. Not even the main disk! 
On 1.85: 
[image]

On 1.9 it's at least a little more helpful: 
[image]

### Version

1.9.2 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Sequoia 15.1.1 (24B91)

### Selected OS

Ubuntu 22.04 (but it is OS independent)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
It doesn't matter what OS I choose, or what model Pi I choose. Once this starts, there is no way to choose a storage medium. I have to reboot the mac in order for it to work again.
```

tdewey-rpi, 2024-12-06:
Thanks for the report, @davidgs 

This appears to be two separate issues:

1) Your system is not allowing storage media to be enumerated, resulting in no drives being listed. 1.9.0 and later include a message for this case.
2) When you used CMD+Q to force terminate the application, this also terminated the DriveListPollThread, which promptly threw a Qt abort.

Regarding (1), please confirm you have granted permission to Raspberry Pi Imager to access data on removable volumes. Without this permission, storage media enumeration will fail.
[image]

Regarding (2), this would appear to be more involved - the termination flow for the DriveListPollingThread doesn't immediately look to be safe to use, and is not triggered in the normal case at all. I may consider revising this at a later point.

tdewey-rpi, 2025-04-30:
Closing as stale

closed 2025-04-30 by tdewey-rpi