raspberrypi/rpi-imager #978: [FEATURE]: USB gadget setup for applicable devices

robiwano, 2024-12-11:
### Is your feature request related to a problem? Please describe.

There does not seem to exist an up-to-date guideline on how to enable USB ethernet gadget for the Raspberry Pi Zero 2 W (for instance). My Windows 10 PC does not detect the USB gadget.

### Describe the solution you would like to see implemented

rpi-imager should have a "USB gadget" tab to setup:
- Network interface (usb_ether)

### Describe alternatives you've considered

Numerous pages and guidelines, none of which I've gotten to work.

### Additional context

N/A

### Version

1.8.5

lurch, 2024-12-11:
See #933

robiwano, 2024-12-12:
Fantastic! Is there any plans for when this will be in a release?

tdewey-rpi, 2024-12-12:
No specific plans at this time - that PR is queued behind some more foundational works in Imager to prepare for 2.0.