raspberrypi/rpi-imager #979: [BUG]: Imager doesn't assign sd card a drive letter after writing (Windows)

michaelbapst, 2024-12-11:
### What happened?

After writing and verifying an image to an SD Card, Imager doesn't assign the new partitions a Drive Letter (C:, D:, E:, etc). The SD Card then doesn't show up in any file manager and isn't accessible from Power Shell/Command Prompt. Current work around is to manually assign a drive letter to the BOOTFS partition using windows Disk Management tool or a 3rd party tool like Minitool Partition Wizard.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Raspberry Pi OS 64

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

michaelbapst, 2024-12-12:
Imager automatically ejects the SD card at the end of verification, so I need to remove and reinsert regardless. Unfortunately the SD card still needs to have a letter assigned to it using one of the methods described above so it shows up in system/file manager.

lurch, 2024-12-12:
> Imager automatically ejects the SD card at the end of verification

There's an option in the Imager settings to turn that off - you might need to press Ctrl + Shift + X to bring up the settings window.

> Unfortunately the SD card still needs to have a letter assigned to it using one of the methods described above so it shows up in system/file manager.

Strange. I don't use Windows myself, but I was under the impression that it would automatically mount the first FAT partition on the SD card when you insert it? Do you have any security software or something that might be preventing this auto-mount? As a test, could you format your SD card to a single FAT32 partition using the "Erase" option in Raspberry Pi Imager, and then see if Windows _does_ auto-mount that when you insert it?

But regardless, if an SD card with Raspberry Pi OS written to it doesn't auto-mount when you remove and then re-insert it, that sounds like a problem with Windows, that RPi Imager will be unable to do anything about? :thinking:

michaelbapst, 2024-12-12:
I will try that after work.

michaelbapst, 2024-12-13:
After trying your suggestion (Using Erase function in Raspberry Imager), The sd card mounted with an assigned letter. I've then tried writing an Official Raspberry PI Image (64 bit Lite) and a custom image, and they both mount correctly with an assigned drive letter. I can no longer reproduce the initial bug (which has been plaguing me for awhile now), and will close this issue. Thanks for your help!

closed 2024-12-13 by michaelbapst