raspberrypi/rpi-imager #980: [FEATURE]: Option to migrate from Raspberry Pi OS with desktop to Raspberry Pi OS Lite

roisec, 2024-12-12:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you would like to see implemented

I have started to use Raspberry Pi OS with desktop, i wondering how i can migrate to Raspberry Pi OS Lite?

### Describe alternatives you've considered

Installing from scratch? 

### Additional context

.

### Version

1.9.2 (Default)

tdewey-rpi, 2024-12-12:
This isn't strictly an issue or request against Imager, and would be best serviced in the Raspberry Pi Forums.

FWIW, I would probably approach this by backing up my useful data, and then installing from scratch - though you may also be able to simply remove packages until you get down to the 'Lite' configuration.

Closing as 'Won't fix', because there's nothing for Imager to do here (without massively broadening the scope of the application).

closed 2024-12-12 by tdewey-rpi

lurch, 2024-12-12:
...and if it's just the "no GUI" feature that you want, you can also configure Raspberry Pi OS with desktop to just boot straight to the console: https://www.raspberrypi.com/documentation/computers/configuration.html#menu-options