raspberrypi/rpi-imager #981: [FEATURE]: Please add under imager settings the ability to enable the serial console

Jibun-no-Kage, 2024-12-17:
### Is your feature request related to a problem? Please describe.

Have to edit the config.txt after imaging to enable UART serial console.  I have many Pi devices that are headless or have a USB-UART board connected to the serial console.  I rarely have Pi device connected to any time of monitor, given the pi devices i have are commercial oriented applications not hobbyist or learning environment based.  I never have a GUI environment.  Would be nice to have SSH enabled as a option on initial boot as well.

### Describe the solution you would like to see implemented

As noted above have imager include settings for UART-USB enable as an option.  Would also be nice of on initial boot SSH could be enabled as well as a configuration option.

### Describe alternatives you've considered

Other than constantly have to edit the config.txt to set enable_UART=1?

### Additional context

See comments above

### Version

1.9.2 (Default)

lurch, 2024-12-18:
> As noted above have imager include settings for UART-USB enable as an option.

Enabling the serial console is _probably_ too much of a niche feature to add a button specifically for it to Raspberry Pi Imager? RPi Imager is designed to be easy enough for complete beginners to use, and every extra option we add adds additional cognitive overhead. If you're using Raspberry Pi OS, then there _are_ already options in `raspi-config` and "Raspberry Pi Configuration" for easily enabling the UART.

> Would also be nice of on initial boot SSH could be enabled as well as a configuration option.

If you're flashing Raspberry Pi OS to your SD card then there **is** already an option to enable SSH - see https://www.raspberrypi.com/documentation/computers/getting-started.html#advanced-options

If you're using an OS other than Raspberry Pi OS, then Raspberry Pi Imager has no idea what steps would be needed to enable SSH in that OS, and so doesn't give you any options.

tdewey-rpi, 2025-01-08:
As @lurch points out, my position on accepting features into Imager is overwhelmingly in favour of the first-time-user experience, to the point where I've excluded a number of features that *I* would find useful.

I find that the hardest features to reject are those that impact fleet devices - and this request would fall into that bracket. There is, however, a solution - respin your OS image to include more advanced customisations, and use that instead of a generic pre-baked OS.

If you're using Raspberry Pi OS, then I'd suggest looking at using `pi-gen` to create a version with your UART configuration baked in.

With regret, feature request rejected.

closed 2025-01-08 by tdewey-rpi