raspberrypi/rpi-imager #982: [FEATURE]: Ignore/Protect Lock Drive

nwgat, 2024-12-21:
### Is your feature request related to a problem? Please describe.

to save people from accidentally wiping their external ssd/hdd etc 
it would be useful to have a ignore or protect lock disk feature

### Describe the solution you would like to see implemented

something like a right click to protect lock the drive

### Describe alternatives you've considered

*

### Additional context

*

### Version

1.9.2 (Default)

lurch, 2024-12-21:
There have been similar requests before (both here and also on the Etcher issues tracker)... the problem becomes that there's no easy way to uniquely identify a removable drive. And even if this "protection" feature _were_ possible, and you rely on it to not overwrite important data, it still doesn't help if you buy a new drive and copy all your data to it and then forget to "protect" it.

It's perhaps a bit of a glib answer, but the safest / simplest / most reliable way is to simply be _very_ careful which drive you select when using Raspberry Pi Imager.
"With great power comes great responsibility" :wink:

tdewey-rpi, 2025-01-08:
As @lurch correctly points out, identifying removable drives is a surprisingly hard problem - one that is trivially accomplished by the user taking care to select the correct drive for what can be a destructive option.

I can't scope this feature, so I'm closing it as 'Won't address'.