raspberrypi/rpi-imager #983: [BUG]: module "kvantum" is not installed

pablingas, 2025-01-02:
### What happened?

rpi-imager 
QQmlApplicationEngine failed to load component
qrc:/main.qml: module "kvantum" is not installed

### Version

1.9.2 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Soaring

### Selected OS

Garuda Linux

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2025-01-02:
> ### What host operating system were you using?
> Other Linux environment
> 
> ### Host OS Version
> Soaring

I did a bit of searching, and "Soaring" appears to be a version of Garuda Linux. However I'm not familiar with this myself (and I suspect that @tdewey-rpi isn't either), so could you please clarify which of the many downloads at https://garudalinux.org/downloads it is that you're actually using? Thanks.

pablingas, 2025-01-02:
Yes, the Gnome Garuda-Genome-Linux-Zen-240428.iso version also uses Arcolinux, Arch, XFCE and the same message appears in terminal when the application begins.

tdewey-rpi, 2025-01-02:
This is a packaging problem for your specific distribution, not an issue with Rpi-imager. Please contact your distribution for support, or try the pre-release AppImage in the releases area (v1.9.0).

Closing as not applicable.

closed 2025-01-02 by tdewey-rpi

uhef, 2025-03-14:
Same issue can be witnessed on Manjaro Linux.

Version available from official Manjaro repositories is `1.9.0-3` at the time of writing.

Starting of the imager fails with:

```
QQmlApplicationEngine failed to load component
qrc:/main.qml: module "kvantum" is not installed
```

Version `1.9.0-4` available from AUR repositories fails similarly.

uhef, 2025-03-14:
I managed to get rpi-imager run on Manjaro by using the ready-built AppImage that one can find on the release page here: https://github.com/raspberrypi/rpi-imager/releases/tag/v1.9.0

abhijeet-singh800, 2025-07-04:
Understood, thanks 💟

obetreyes6, 2025-10-05:
> Workaround => If you change the Qt theme to anything else it starts working. Hope it helps

it works for arch linux 

- launch qt5ct  - "qt5ct"
- change the style  to Fusion under the Appearance tab  
- click apply then ok
- then set the env variable   "echo 'export QT_STYLE_OVERRIDE=Fusion' >> ~/.bashrc
source ~/.bashrc"
- run rpi imager