raspberrypi/rpi-imager #984: [BUG]: Blank Device and OS tabs if corrupted SD is mounted

HitLuca, 2025-01-02:
### What happened?

I recently installed rpi OS on an SD card, inserted it into my raspberry and it was corrupted. So I plugged it back in my PC and started raspberry pi imager: this time both the device and os tabs were blank, suggesting some kind of issue with the program. Reinstalling didn't help, and everything got back working as soon as I formatted the SD card from Windows. 

### Version

1.8.5

### What host operating system were you using?

Windows

### Host OS Version

Windows 10

### Selected OS

RPI OS 64bit

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2025-01-06:
Thanks for the report, @HitLuca 

I'd expect the 'Device' and 'OS' windows to be populated regardless of the state of the 'Media' window.

Your report implies a network failure - as the only critical path for this data is a network fetch and a UI update. It's also possible that there's an undocumented dependency between the UI update and the media detection. I'll attempt to reproduce this failure to confirm.

HitLuca, 2025-01-07:
My guess is that the imager fails to load up the media devices due to a weird/corrupt partition on the sd card, and this breaks the rest of the population steps. I have thrown away the faulty sdcard so I cannot reproduce the issue, but happy to help out in some other ways

lurch, 2025-01-07:
> I have thrown away the faulty sdcard

So the card became corrupted **again**? In your earlier message you said "everything got back working as soon as I formatted the SD card".

HitLuca, 2025-01-07:
@lurch sorry I misworded my first message. The sdcard was still faulty as in the raspberry couldn't boot off of it (and the data verification process failed after writing), but the imager itself worked (all panels loaded properly) if the card was freshly formatted

lurch, 2025-01-07:
Ahhh, it might have been a "fake capacity" microSD card? https://rmprepusb.com/tutorials/007-all-about-fake-sd-cards-and-usb-flash-drives/

If it failed verification, then it's not much of a surprise that the Raspberry Pi failed to boot from it! :joy:

HitLuca, 2025-01-07:
Could totally have been, but i would expect that the imager shouldn't have issues populating anyway

lurch, 2025-01-07:
> Could totally have been, but i would expect that the imager shouldn't have issues populating anyway

I entirely agree - sorry if I accidentally implied otherwise.

closed 2025-08-01 by tdewey-rpi