raspberrypi/rpi-imager #986: [FEATURE]: I can't code C, but add bazzite! (please)

weblabsaus, 2025-01-05:
### Is your feature request related to a problem? Please describe.

I can't find Bazzite Arm64 iso too.

### Describe the solution you would like to see implemented

Add for Raspberry pi 5

### Describe alternatives you've considered

I dunno, Fedora didn't work

### Additional context

(bazzite)[https://bazzite.gg]

### Version

1.9.2 (Default)

lurch, 2025-01-05:
Are you one of the maintainers of Bazzite? It's the OS-maintainer that needs to request RPi Imager support for their OS, as it also relies on a JSON file hosted on their website.

Hmmm, https://bazzite.gg/#image-picker mentions nothing about Raspberry Pi??

weblabsaus, 2025-01-06:
no i don't maintain it

tdewey-rpi, 2025-01-06:
This isn't a feature request - and there's no sign of Bazzite supporting any Raspberry Pi Hardware. I'd suggest contacting that project about support.

Closing as 'Won't Fix', because there's nothing for Imager to do here.

closed 2025-01-06 by tdewey-rpi