raspberrypi/rpi-imager #988: [BUG]: Long Filename Save Bug on FAT Partition in Windows

mgordel, 2025-01-07:
### What happened?

In my fork of the project (https://github.com/Web3-Pi/web3-pi-imager), I implemented functionality to save an empty file named `format_storage` on the image being modified by the `DownloadThread::_customizeImage()` function. 
In my case (fork) this functionality is triggered when the user selects the appropriate checkbox in the GUI. The code responsible for saving the file is as follows:

```cpp
fat->writeFile("format_storage", "");
```

## Problem 
On macOS and Linux, the file is saved correctly as `format_storage`. However, on Windows 10, the saved file name is incorrect: `format_s.tor`

## Debugging

I investigated the issue and identified that the problem most likely occurs in the function:

```cpp
DeviceWrapperFatPartition::getDirEntry()
```

Specifically, in the `createIfNotExist` condition (402 line), where the current implementation for creating long filenames on Windows 10 appears to be incorrect.

## Steps to Reproduce
 1. Add the following line: ```fat->writeFile("format_storage", "");``` in the file `downloadthread.cpp` in function `DownloadThread::setImageCustomization` before line #996
 2. Perform the image customization process.
 3. Check the file name on the modified image:
    - On macOS/Linux: The file is saved correctly as `format_storage`.
    - On Windows 10: The file is saved as `format_s.tor`.
 
## Probable Cause of the Bug

The issue arises due to differences in how Windows 10 handles character encoding and data access depending on the type of strings used (ANSI vs. UTF-16). Windows requires UTF-16 for most of its file-related APIs, whereas the current implementation appears to use UTF-8 or another encoding, leading to incorrect file name handling.

## Proposed Solution

Replace the following code ([devicewrapperfatpartition.cpp](https://github.com/raspberrypi/rpi-imager/blob/b01593af9a00f0ba7671c0b55eb0d134f03e6188/src/devicewrapperfatpartition.cpp#L441)):

```cpp
QString longFilenameWithNull = longFilename + QChar::Null;
char *longFilenameStr = (char *) longFilenameWithNull.data();
int lfnFragments = (longFilenameWithNull.length()+12)/13;
int lenBytes = longFilenameWithNull.length()*2;

/* long file name directory entries are added in reverse order before the 8.3 entry */
for (int i = lfnFragments; i > 0; i--)
{
    memset(&longEntry, 0xff, sizeof(longEntry));
    longEntry.LDIR_Attr = ATTR_LONG_NAME;
    longEntry.LDIR_Chksum = shortFileNameChecksum;
    longEntry.LDIR_Ord = (i == lfnFragments) ? lfnFragments | LAST_LONG_ENTRY : lfnFragments;
    longEntry.LDIR_FstClusLO = 0;
    longEntry.LDIR_Type = 0;

    size_t start = (i-1) * 26;
    memcpy(longEntry.LDIR_Name1, longFilenameStr+start, qMin(lenBytes-start, sizeof(longEntry.LDIR_Name1)));
    start += sizeof(longEntry.LDIR_Name1);
    if (start < lenBytes)
    {
        memcpy(longEntry.LDIR_Name2, longFilenameStr+start, qMin(lenBytes-start, sizeof(longEntry.LDIR_Name2)));
        start += sizeof(longEntry.LDIR_Name2);
        if (start < lenBytes)
        {
            memcpy(longEntry.LDIR_Name3, longFilenameStr+start, qMin(lenBytes-start, sizeof(longEntry.LDIR_Name3)));
        }
    }

    writeDirEntryAtCurrentPos((struct dir_entry *) &longEntry);
}
```

With:
```cpp
QString longFilenameWithNull = longFilename + QChar::Null;
char *longFilenameStr = (char *) longFilenameWithNull.utf16();
int lenBytes = longFilenameWithNull.length() * 2;
int lfnFragments = (lenBytes + 25) / 26;

/* long file name directory entries are added in reverse order before the 8.3 entry */
for (int i = lfnFragments; i > 0; i--)
{
    memset(&longEntry, 0xff, sizeof(longEntry));
    longEntry.LDIR_Attr = ATTR_LONG_NAME;
    longEntry.LDIR_Chksum = shortFileNameChecksum;
    longEntry.LDIR_Ord = (i == lfnFragments) ? (LAST_LONG_ENTRY | i) : i;
    longEntry.LDIR_FstClusLO = 0;
    longEntry.LDIR_Type = 0;

    size_t start = (i-1) * 26;
    memcpy(longEntry.LDIR_Name1, longFilenameStr+start, qMin(10U, lenBytes-start));
    start += 10;
    if (start < lenBytes)
    {
        memcpy(longEntry.LDIR_Name2, longFilenameStr+start, qMin(12U, lenBytes-start));
        start += 12;
        if (start < lenBytes)
        {
            memcpy(longEntry.LDIR_Name3, longFilenameStr+start, qMin(4U, lenBytes-start));
            start += 4;
        }
    }

    writeDirEntryAtCurrentPos((struct dir_entry *) &longEntry);
}
```

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 10 Pro

### Selected OS

Raspberry Pi OS 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

lurch, 2025-01-07:
Maybe a duplicate of (or at least closely related to) #945 ?

maxnet, 2025-01-07:
- Note that some of the changes you make have no impact.
E.g. you change a call to QString's data() function to utf16()
But for ordinary strings, utf16() does nothing more than call data()...

https://github.com/qt/qtbase/blob/dev/src/corelib/text/qstring.cpp#L7049

- Your changes to lenBytes and lfnFragments calulcation also should make no difference.

- If replacing sizeof(longEntry.LDIR_Name3) for a numeric value does make a difference that however could be a sign that  the compiler (version) you (and Tom?) are using nowadays to compile Imager, is behaving different from the one I used when writing the original code, and that do is something you may want to investigate.

I do remember that the Mingw version I used at the time, was also behaving a bit oddly in that field.
I initially marked the structs with "__attribute__ ((packed))" hoping that would disable memory alignment of the fields in the struct.
Worked fine on Linux but not on Mingw/Windows.
I changed it to using "#pragma pack(push, 1)" instead in this commit: https://github.com/raspberrypi/rpi-imager/commit/5fa3fbe8dcef4ab01c0a2fed5638759265c3f7f6#diff-3c2bf5c911d6ad07dc7e40ede5210004b6a24785d564779c8349fa0bae96d842
And that did work for me with Mingw as well.
If it no longer does in the compiler verison you are using, than you may need to find another option.

You may also want to try adding a debug print statement allong the lines of:

qDebug() << "longEntry sizes according to compliler (should be 10 12 4)" << sizeof(longEntry.LDIR_Name1) << sizeof(longEntry.LDIR_Name2) << sizeof(longEntry.LDIR_Name3);

To see if the compiler is indeed behaving oddly in this field.

closed 2025-01-08 by tdewey-rpi

mgordel, 2025-01-09:
Yes, @maxnet you are right, sizeof is not a problem. When fixing the bug I tried to check all aspects and I guess I was too fast with those constant numbers.

I debugged the code again using your hint with `qDebug() << "longEntry sizes according to compiler...` and indeed there is no difference.

The only significant error is `longEntry.LDIR_Ord` and it was the one causing the problem.

So you are right. I prepared another PR: https://github.com/raspberrypi/rpi-imager/pull/992 with the reverted changes for sizes because this way will indeed be safer.

I would leave the changes with `longFilenameStr` and `lfnFragments` in their new form (in my opinion they improve readability)