raspberrypi/rpi-imager #994: [BUG]: Download error

GeekWizkid, 2025-01-12:
### What happened?

[image]

### Version

1.8.5

### What host operating system were you using?

Windows

### Host OS Version

Windows Feature Experience Pack 1000.19060.1000.0

### Selected OS

Ubuntu Desktop 24.10 (64 bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2025-01-13:
Thanks for the report, @GeekWizkid.

In this case, it appears the Canonical server is reporting a Bad Request - which is deeply unexpected, because Imager is routinely used to create Ubuntu media.

Do you see this message on every attempt to use Ubuntu? And does it appear on other OSes from the list?

GeekWizkid, 2025-01-13:
I've been trying to install Ubuntu for a week now, tested all available versions with no success. However, Kali installed just fine.

tdewey-rpi, 2025-01-13:
Thanks for the rapid response.

As you're only seeing this with Ubuntu, I'd suggest your best bet is to contact Canonical. I'm pinging @waveform80 in case they're aware of any particular problems.

Otherwise, it would appear that there's something afoul in your internet connection. Typical remedies involve power cycling your modem/router, and contacting the ISP. One thing that may also be worth trying is downloading the image manually:

Server 24.10:
http://cdimage.ubuntu.com/releases/oracular/release/ubuntu-24.10-preinstalled-server-arm64+raspi.img.xz

Desktop 24.10:
http://cdimage.ubuntu.com/releases/oracular/release/ubuntu-24.10-preinstalled-desktop-arm64+raspi.img.xz

And then using the 'Use Custom' option in the OS selector to select the file manually.

GeekWizkid, 2025-01-16:
I tried putting it in manually. When booting, green uniform background (graphic mode) and that's it.

lurch, 2025-01-16:
You might want to verify the sha256sum of your image, to check that it downloaded correctly? https://cdimage.ubuntu.com/releases/oracular/release/SHA256SUMS