raspberrypi/rpi-imager #995: [BUG]: Imager caches the drive size and reports too small when a larger drive is inserted

Matthew1471, 2025-01-16:
### What happened?

Hardly a big bug but a bug nevertheless.

Tried to use Imager to flash the Pi OS Lite (64-bit) to a 2.0GB card and was told it needs to be 6.0 GB or greater.. took out the 2.0GB card and inserted a 64 GB card and was still told I need a card 2.0 GB or larger.. clicked on the storage again where it said D:\ is 2.0 GB think I had to open and close the storage screen numerous times before it would refresh.

The constraint check should probably check it has the most up to date drive information.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

11 24H2

### Selected OS

Raspberry Pi OS 64-bit Lite

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Not seeing any logs in C:\Users\Matthew\AppData\Local\Raspberry Pi\Imager.. should I be looking somewhere else for logs?
```

lurch, 2025-01-16:
> Tried to use Imager to flash the Pi OS Lite (64-bit) to a 2.0 GB card and was told it needs to be 6.0 GB or greater.. took out the 2.0 GB card and inserted a 64 GB card and tried to "Next" and was still told I need a card 2.0 GB or larger.. 

Hmmm, regardless of whether the "Choose Storage" dialog has trouble updating the drive information or not (which it obviously shouldn't 😉 ), it feels to me that if you remove the currently-selected card then Imager should "forget" that drive-selection, and so clicking "Next" should prompt you to go back and choose a different drive from the "Choose Storage" dialog?

Can you provide more details about the exact models of micro SD cards that you're using please, and more details about the "internal reader"? (e.g. if it's an internal reader in a laptop, which make & model of laptop?) If you have a USB SD card-reader available, could you see if that behaves any differently if you try to repeat the above scenario?

Matthew1471, 2025-01-16:
I booted my 64 GB card and realised I'd accidentally selected the full rather than lite (hence the 6.0 GB warning instead of 2.6 GB warning).. have reproduced again.. but I can go one better - I can remove the SD card have it disappear from the Storage popup, have the main screen now say "SD CARD" instead and click "Next" to still be told:

"Would you like to apply OS customisation settings?"

..to which I say "Yes" or "No" to then be asked:

"All existing data on 'SD Card' will be erased.
Are you sure you want to continue?"

..to which I say "Yes" (what SD card?) to be told:

"Storage capacity is not large enough.
Needs to be at least 2.8 GB."

There is nothing in the drive and even Raspberry Pi Imager knows this :)

Laptop is a "Dell XPS 13 9305" with an embedded "Realtek PCIE CardReader" (according to device manager). I don't think this is card reader specific however as the Storage screen is updating absolutely fine showing nothing inserted yet the cache still prevails. The SD card is a "Nokia" 2.0 GB microSD from decades ago.. I don't suppose this truly matters either considering I can put the "right" card in and still get the message.

tdewey-rpi, 2025-01-22:
Thanks for the report, @Matthew1471 - in particular, taking the time to clearly capture some reproduction steps.

I agree this is an undesirable behaviour, and will take a look once I get back around to Imager duties.

PR #1156 (2025-08-01): 1.9.6: Fix roll-up
- Added conditional ejecting of the disk based on the _ejectEnabled flag in DownloadExtractThread.
- Implemented saving of miscellaneous options (beep, eject, telemetry) to persistent storage in OptionsPopup.

Fixes #1154 
Fixes #1089
Fixes #1148 
Fixes #1152 
Fixes #995 
Fixes #1111
closed 2025-08-01 by tdewey-rpi