raspberrypi/rpi-imager #996: [FEATURE]: Option to Upload A Completely Custom Raspberry Pi Cloud Init File

kevinlutzer, 2025-01-19:
### Is your feature request related to a problem? Please describe.

For both personal use and previous jobs, I have had to flash multiple Raspberry PIs (in a cluster or just single use) and then use a tool like Ansible to initialize them with the Debian packages, network config, and Ubuntu Snaps I needed for their desired purpose. This can be very tedious and it would be nice if the RPi Imager tool could further initialize the OS.

### Describe the solution you would like to see implemented

Currently, the Raspberry Pi imager tool allows configuring things like the password, username, hostname, wifi networking setup, and timezone via cloud-init. It would be nice to have the option to upload a completely custom cloud-init config file. During the flashing process, the tool would copy the custom cloud-init file onto the target OS. 

This would be a massive workflow improvement for me. Now, I wouldn't have to use different workflow automation tools to install and set up additional system dependencies after the RPi has been booted properly. Note that tools/services like Proxmox, and Azure VM allow you to upload a custom cloud-init file when imaging a VM. 

### Describe alternatives you've considered

N/A

### Additional context

N/A

### Version

1.9.2 (Default)

maxnet, 2025-01-19:
It is currently not exposed in the GUI, but when starting Imager from the command-line like "rpi-imager --cli https://server/imagefile /dev/sdX" you do have "--cloudinit-userdata customfile" and "--cloudinit-networkconfig customfile" options.

kevinlutzer, 2025-01-19:
Very cool! I didn't realize that. What would it take to build out that feature so it is supported by the GUI? In the OS Customization -> Edit settings pane, would it be possible to add an `Advanced Features` dropdown that would expose two uploads? One for a user-data file and another for the network-config file?

lurch, 2025-01-20:
I hope you don't mind me asking, but why are the CLI options not sufficient for your use-case?

curtisgalloway, 2025-05-21:
Not the OP, but chiming in here.  In my use case, I want to configure a Pi to disable wifi and only have a network connection over USB to the host, so any extra packages must be loaded directly onto the image.  I only wanted to write a few files, so I was able to just use `--cli --first-run-script` to include the files as a base64-encoded tarball directly in the [script file](https://gist.github.com/curtisgalloway/919b5f5fb330586261d3d5d156ae8735) and unpack them at first run.  For bigger files, using that together with `--disable-eject` would allow copying over more files to the boot partition that can be processed with the `first-run.sh` script.

lurch, 2025-05-21:
> In my use case, I want to configure a Pi to disable wifi and only have a network connection over USB to the host, so any extra packages must be loaded directly onto the image.

I believe our "recommended" way of doing that would be to build a custom image using rpi-image-gen - see https://www.raspberrypi.com/news/introducing-rpi-image-gen-build-highly-customised-raspberry-pi-software-images/

curtisgalloway, 2025-05-21:
Good idea - that's a lot less hacky; I'll try that next.