raspberrypi/rpi-imager #997: [BUG]: Scrolling is extremely slow in dropdown boxes.

st19311, 2025-01-23:
### What happened?

Scrolling in the dropdown boxes only works very slowly, no matter how fast I spin the wheel on my mouse. It almost seems like it's debouncing the events. 

This makes it very time consuming if I want to, for example, select "de" in the keyboard layout, because I have to go aaaaall the way up from "us". 
It doesn't really help that the dropdown boxes don't appear have any basic quality-of-life functionality - the scroll bar on the side is just visual and when I click, it's as if I'm dragging down the menu (so to go up, I have to drag down), and I can't, for example, press "D" to scroll up to D, nor can I press "home" to go all the way to the top. I can't even say that it feels like 1995 again, because all that actually worked in Windows 95 lol. My only options are the slow scroll, or pulling down the menu repeatedly until I'm at the top (which is actually faster than scrolling).

I realize that I can workaround that by just typing into the textbox what I need to type in, but when I don't know it yet, because it's the first time I'm scrolling up, it's pretty tedious.

### Version

1.8.5

### What host operating system were you using?

Windows

### Host OS Version

Windows 11

### Selected OS

Raspberry Pi OS Bookworm 64 Bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-01-23:
See also: #558 #598 #865

tdewey-rpi, 2025-05-01:
This is resolved in the 1.9-series releases, which we're moving to publish on the website in the near future.

tdewey-rpi, 2025-05-01:
Closing as resolved, as 1.9 is already released in the Releases area.

closed 2025-05-01 by tdewey-rpi

st19311, 2025-05-05:
> This is resolved in the 1.9-series releases, which we're moving to publish on the website in the near future.

Awesome job, thanks.