raspberrypi/rpi-imager #998: [BUG]: No OS options for 3D Printing with Raspberry Pi 5 selected.

GySlaughter, 2025-01-24:
### What happened?

Imager 1.8.5 shows no OS options for 3D Printing other than Repetier-Server when Raspberry Pi 5 selected.  All Options are available when Raspberry Pi 4 is selected.

### Version

1.8.5

### What host operating system were you using?

Windows

### Host OS Version

22H2 Build 19045.5371

### Selected OS

Not able to select desired OS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-01-24:
I believe this is a "problem" that those third-party 3D-printing OSes would need to resolve, this isn't a problem of RPi Imager.
See e.g. https://octopi.octoprint.org/rpi-imager.json or https://os.mainsail.xyz/rpi-imager.json where they don't list Pi 5 in their `devices` list.

tdewey-rpi, 2025-02-06:
Thanks for the report, @GySlaughter.

As @lurch correctly points out, no other 3D Printing OS has marked their OS as compatible with Raspberry Pi 5. I don't have any direct control over this labelling - so I'd encourage you to contact the OS you'd like to use and ask them to look at supporting Raspberry Pi 5-class hardware.

Closing as this isn't an issue with `rpi-imager`, but with OS support offered by third parties.

ammgws, 2025-05-06:
FYI @GySlaughter OctoPi now supports RPi5 https://github.com/guysoft/OctoPi/releases/tag/1.1.0