raspberrypi/rpi-imager #1002: [BUG]: Unable to use image compressed using zstd on Windows

yodaldevoid, 2025-02-10:
### What happened?

When trying to flash an image compressed using `zstd` on Windows, I got the error below. `zstd` is installed on the system, available on the path, and I can run it manually with no issue.

### Version

1.9.0

### What host operating system were you using?

Windows

### Host OS Version

24H2

### Selected OS

Custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Error extracting archive: Can't initialize filter; unable to run program "zstd -d -qq"
```

slokhorst, 2025-02-26:
Same issue here. This worked previously.

lurch, 2025-03-20:
@tdewey-rpi This looks suspiciously similar to https://github.com/libarchive/libarchive/issues/2488 ?

yodaldevoid, 2025-03-20:
Agreed, though it doesn't state why libarchive can't find the system utilities for the fallback method.

I suppose the true solution is to compile in support.

tdewey-rpi, 2025-03-20:
Good catch, @lurch.

This does indeed look like something upstream have seen - I'll rev our vendored copy and see if we can persuade it to use included code rather than calling out to the system

tdewey-rpi, 2025-06-02:
Fixed in $HEAD, 1.9.4 in internal testing with release (hopefully) imminent.