raspberrypi/rpi-imager #1005: [BUG]: RPI imager 1.9.0 just disconnected from internet

Mtepihasti5, 2025-02-17:
### What happened?

I was using RPI imager 1.9.0 and it just said that it cannot resolve github.com
I check my wifi. My laptop was still connected to wifi.
I just tried again and IT STARTED FROM THE BEGGINING.
Please for the love of god help me.

### Version

1.9.0

### What host operating system were you using?

Windows

### Host OS Version

Windows 11 24H2 (i think)

### Selected OS

Retro pie 4.8 for rpi 3 and others

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

USB SSD

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-02-17:
Duplicate of #1006 

If your internet connection is so unreliable that Raspberry Pi Imager is unable to download your required image file in one go, then I guess you could download the image file on your laptop using some other program that _does_ support resuming interrupted downloads, and then use RPI Imager's "Use custom" option to write your already-downloaded image file.

Mtepihasti5, 2025-02-18:
i use fdm but i am going to close issue
And i downloaded it using fdm (the image file)

closed 2025-02-18 by Mtepihasti5