raspberrypi/rpi-imager #1006: [FEATURE]: Add continue from last time

Mtepihasti5, 2025-02-17:
### Is your feature request related to a problem? Please describe.

I was downloading retropie 4.8 for rpi 3 and others.
I did see that it can't resolve github.com
And i had to start from the beginning.
Then i said:
Aw BS.
So anyone please help me.

### Describe the solution you would like to see implemented

It should continue from when it was cancelled.
On the start it should ask me should it continue from when it was cancelled

### Describe alternatives you've considered

I'd consider that it tries until it can continue

### Additional context

Please help me

### Version

1.9.0

lurch, 2025-02-17:
Unfortunately looking at https://github.com/raspberrypi/rpi-imager/blob/qml/src/imagewriter.cpp#L168 and https://github.com/raspberrypi/rpi-imager/blob/qml/src/imagewriter.cpp#L421 it doesn't actually store the URL of the file you were trying to download, but _only_ the SHA256 checksum of the file that you last downloaded.

So on a SUCCESSFUL download & write attempt, the `lastdownload.cache` file will be updated with your (complete) image file, and the `lastDownloadSHA256` setting will be updated with the checksum of your (complete) file. So the next time you try to download & write the same image, RPi Imager asks itself "Does the `lastDownloadSHA256` match the checksum of the image I'm being asked to download? Oh, it does so I'll just write `lastdownload.cache` to the SD card instead."

But on an UNSUCCESSFUL download & write attempt (e.g. if your internet connection gets interrupted) the `lastDownloadSHA256` setting will be updated with the checksum of your incomplete file. So the next time you try to download & write the same image, RPi Imager asks itself "Does the `lastDownloadSHA256` match the checksum of the image I'm being asked to download? No it doesn't, so I'll ignore the `lastdownload.cache` file and download the image instead."

TLDR: This won't be an "easy fix".

Mtepihasti5, 2025-02-18:
it's all right thati it isn't an easy fix but please help me

Mtepihasti5, 2025-03-16:
Do i close it or not

tdewey-rpi, 2025-05-01:
Feature request (regretfully) rejected.

I can absolutely understand the pain of this - in my younger years I recall the horrors of downloading content on unstable connections.

Unfortunately, there still isn't, I believe, a decent and reliable solution for _HTTPS_ downloads. There's some best-effort mechanisms around ranged-fetches, session resumption and so on, but nothing that compared with something like a BitTorrent or a gossip protocol.

Why is that relevant? Because all of our OS vendors host over HTTP/S.

You could, however, try to drive it manually: Get the URL of the OS you're trying to download, and use a browser to fetch it. The browser will use best-effort mechanisms to try and get the file. Then, in rpi-imager, treat it as flashing a 'Custom OS'.

Separately, if you're paying for connectivity and finding it lacking, I'd encourage you to follow up with your connectivity provider. Your mileage may vary, but you might also be helping them find a problem they hadn't recognised.

closed 2025-05-01 by tdewey-rpi

Mtepihasti5, 2025-05-03:
DO NOT REJECT

Mtepihasti5, 2025-06-25:
OH CRAP MAN

napisao je: