raspberrypi/rpi-imager #1007: [FEATURE]: Allow user to download from other location

fckwan, 2025-02-19:
### Is your feature request related to a problem? Please describe.

Please allow user to choose and download the OS from a mirror location.

### Describe the solution you would like to see implemented

Please allow user to choose and download the OS from a mirror location.
It can diversify the load to other mirror site. Lower the distance the OS image need to travel.

### Describe alternatives you've considered

User can download the image themselves and then use the imager to write to sd card.

### Additional context

N/A

### Version

1.9.2 (Default)

lurch, 2025-02-19:
Raspberry Pi Imager is designed to be easy for complete beginners to use, and I suspect that allowing users to specify a mirror location would just make it more confusing for them? (e.g. a complete beginner wouldn't even know what a "mirror location" is, and even an experienced beginner probably wouldn't have any idea how to figure out what mirror locations are available for the particular OS that they want to use)
IMHO of course.

fckwan, 2025-02-19:
This option can be hidden in the setting or in advanced settings.
Most user do not need to worry about this.

tdewey-rpi, 2025-05-01:
Feature request rejected.

This is already somewhat-trivially implementable using existing primitives.

For people who want to host an alternative OS list, that function already exists: you specify it via a program argument.

For people who want to load-balance and geo-locate, this is a conventional web architecture problem that you fix on the server, not in the application.

closed 2025-05-01 by tdewey-rpi