raspberrypi/rpi-imager #1010: [BUG]: --cli requires OpenGL libs to be installed

graham2071, 2025-02-22:
### What happened?

Running rpi-imager with `--cli`` on Raspberry Pi OS Lite (with no desktop) fails because of missing OpenGL libraries.

```console
$ sudo rpi-imager --cli --first-run-script ./firstrun.sh 2024-11-19-raspios-bookworm-arm64-lite.img.xz /dev/nvme0n1
/tmp/.mount_rpi-imjBVGEC/AppRun.wrapped: error while loading shared libraries: libGLX.so.0: cannot open shared object file: No such file or directory
```

I would expect the `cli` mode to work without those libs.

**Workaround**

I've installed additional packages one by one until it works:

```bash
sudo apt install libglx0
sudo apt install libopengl0
sudo apt install libegl1
sudo apt install libgl1-mesa-glx
```

That surely are dependencies and maybe only the last is needed.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Raspberry Pi OS Lite (64 bits)

### Selected OS

Raspberry Pi OS Lite (64 bits)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-02-24:
See more information in #961

tdewey-rpi, 2025-04-30:
A regrettable issue, however I'm going to have to close this as a duplicate of #961, as it's fundamentally the same issue: 'cli mode' is a flag against the GUI application, and as a result you must have GL libraries (and other GUI-supporting libraries) present to satisfy the dynamic linker.