raspberrypi/rpi-imager #1019: [FEATURE]: Tracking (non-x)wayland support

jedahan, 2025-03-07:
### Is your feature request related to a problem? Please describe.

Is there a place where we could track the upstream dependency issue described in https://github.com/raspberrypi/rpi-imager/blob/qml/appimagecraft.yml#L24-L32 ? It would be for distro package maintainers who cannot use AppImage to support users who cannot use XWayland to keep abreast of whats needed.

### Describe the solution you would like to see implemented

Share a link to wherever the work being done to fix the dependency issues are. Links could be in AppImage or Wayland or Qt or somewhere else, just guessing here.

### Describe alternatives you've considered

Create a tracking issue for wayland support in this repository, and clearly mark it as "This is just a tracking issue, work needs to be done by external contributors, not by rpi foundation". And then maintainers can just follow that one.

### Additional context

I dunno, its a hard problem, thanks for your consideration.

### Version

1.9.2 (Default)

lurch, 2025-03-08:
There's additional notes in #942 an #975 that you might find useful / informative.

closed 2025-03-08 by jedahan

lurch, 2025-03-09:
> It would be for distro package maintainers who support users who cannot use XWayland to keep abreast of whats needed.

Just out of personal curiosity, **are** there distros where Wayland is supported, but XWayland isn't?

jedahan, 2025-03-12:
Apologies for the late response - I haven't tested recently, but I used to maintain some packages for kisslinux. In it, users can install wayland and sway without xwayland. I will test in a vm to see if this is still an option or if the latest default of the package includes xwayland.

jedahan, 2025-03-12:
So to be as clear as I can, there are two levels of support here
1) the distro supporting XWayland - if the distro supports xwayland, then users are not blocked from running rpi-imager because they can install XWayland to use it. XWayland is a runtime dependency of rpi-imager, this is normal.

2) the distro supports those who want to run wayland without xwayland - if there is a way for the distro to support their users in this config, they cannot use rpi-imager, this is normal but would be nice to support user choice if there is a path forward. Not sure if that path starts in AppImage project (providing a way to be more specific about ABIs needed), or rpi-imager project (some supported build options, as seen in the commented out build instructions), or wayland (pushing for better ABI guarantees), or somewhere else.