raspberrypi/rpi-imager #1020: [BUG]: rpi-imager 1.8.5 on Fedora does not use the whole SDcard

compata, 2025-03-14:
### What happened?

I have previously used rpi-imager on Fedora to setup an RPI 5 and found that the Linux partition created used the entire SDcard aside from the 512 MB used by the FAT32 partition. However, when using version 1.8.5 to create a new 128 GB SDcard, the Linux partition is consistently allocated only about 5 GB. Of course, I know how to expand it but that shouldn't be necessary.

The only customization was specifying the wifi network and password. I presume the PiOS version installed was the latest as rpi-imager obtains it online.

### Version

1.9.2 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Fedora x86_64 version 41

### Selected OS

Raspberry Pi OS 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-03-14:
This isn't a bug with Imager - "all" it does is copy the disk image onto your SD card. Any partition-resizing is the responsibility of the OS, not of Raspberry Pi Imager, e.g. on Raspberry Pi OS the partitions get resized to use up the entire SD card the first time you boot it up.

lurch, 2025-03-14:
Ah, I've just realised that I misread your original comment - I thought you were saying that you had flashed Fedora to your SD card, and it was **that** that wasn't getting resized to use up your whole SD card.
But if you've actually flashed Raspberry Pi OS to your SD card and RPi OS hasn't resized to use your whole card, then you may have discovered a bug in RPi OS's resize-script, so pinging @XECDesign
@compata Can you please provide more details about the particular microSD card that you're using please?

compata, 2025-03-16:
I withdraw my concern. I had not previously enlarged the initial partition size and did not realise that the Pi's initialization did that. I had never previously looked at the disk layout before first using the new SD card.

closed 2025-03-16 by compata