raspberrypi/rpi-imager #1025: [FEATURE]: Add diet pi

godalming123, 2025-03-22:
### Is your feature request related to a problem? Please describe.

d

### Describe the solution you would like to see implemented

d

### Describe alternatives you've considered

d

### Additional context

d

### Version

1.9.2 (Default)

lurch, 2025-03-24:
It's up to the Diet Pi maintainers to request inclusion in https://downloads.raspberrypi.com/os_list_imagingutility_v4.json and there's some Google-form that they have to fill out to do that. I can never remember that URL, but hopefully @tdewey-rpi will be able to provide it... 😉

nafanz, 2025-03-30:
Apparently we are talking about this link:
https://forms.gle/X7HpED5fYcjW1k8c7

Her mention was noticed here:
https://github.com/raspberrypi/rpi-imager/issues/941#issuecomment-2428925180

MichaIng, 2025-04-19:
We were digging into it and have a JSON which seems to work and which we can host/maintain/update on our server.

Here the full example for testing, being aware that only the `os_list` part is needed: https://github.com/MichaIng/DietPi/pull/7450/files

MichaIng, 2025-06-12:
@tdewey-rpi 
Sorry for the ping. Just avoiding to split discussion across another duplicate request. As mentioned above, password change is now mandatory in DietPi, JSON hosted and updated automatically, Google form was filled before that change, Let me know if something is missing.

I already wanted to open a PR, but as far as I can see, the Imager fetches the OS list from https://downloads.raspberrypi.com/os_list_imagingutility_v4.json, so no source code change required.

nafanz, 2025-12-12:
@tdewey-rpi 
Please tell me, is there any news on this issue?

As far as I understand, @MichaIng provided all the data

MichaIng, 2026-01-01:
Thank you very much, DietPi can now be selected directly from the Raspberry Pi Imager 👍.