raspberrypi/rpi-imager #1026: [BUG]: Can't flash any image

komorebithrowsatable, 2025-03-24:
### What happened?

Hi! Yeah, It's a known issue, just letting you know once again

[image]

### Version

1.9.2 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Fedora

### Selected OS

Fedora

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-03-24:
> ### Version
> 1.9.2 (Default)

That's funny, your screenshot says "Raspberry Pi Imager v1.8.5" ?

> ### Host OS Version
> Fedora

There are a several 3rd-party "re-packagings" of Raspberry Pi Imager for Linux. Could you please clarify exactly where you downloaded Raspberry Pi Imager from (and in what format), and how you installed it on Fedora? Thanks.

lurch, 2025-03-24:
> The problem is caused by selinux...

Right, see https://github.com/raspberrypi/rpi-imager/issues/396#issuecomment-1930312124

komorebithrowsatable, 2025-03-24:
Yeah, that's what I'm talking about. The page clearly says "Linux", and selinux is a kernel feature. There's a huge difference between supporting linux and supporting a single distro. 

[image]

tdewey-rpi, 2025-04-30:
Closing this one as 'Won't fix'.

Fedora and RHEL have long taken a particularly aggressive security posture - enforcing SELinux out of the box. In one sense, that's to be commended.

However, I cannot dedicate time to supporting specific distros and their baroque requirements. I am, however, happy to accept a PR for a SELinux file.

Finally, I absolutely cannot support third-party distributions of the application - it's too onerous to track the various modifications and unique failures these may introduce.

closed 2025-04-30 by tdewey-rpi