raspberrypi/rpi-imager #1028: [BUG]: Etcher doesn't set password

BR0KK, 2025-03-30:
### What happened?

I've tried various versions of this program wit various distros and versions of these distros. The flasher does not set the password for the first login correctly. I even tried older versions of this program. Nothing works!

The user is created tho. 

I'm trying a headless install of either ubuntu server or pi OS lite. I have no KB and mouse. If i plug the r-pi (power and network) it boots. SSH is enabled (Port is open) i can even connect to the r-pi but i always get "access denied" on any password i've set or even the defaults. 

### Version

1.9.2 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Sonoma

### Selected OS

ubuntun server 24.xxx

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
there are no logs as i cant get into the pi withut logging in first to read the logs ....
```

lurch, 2025-03-30:
> Etcher doesn't set password

???

> I'm trying a headless install of either ubuntu server or pi OS lite. I have no KB and mouse. If i plug the r-pi (power and network) it boots. SSH is enabled (Port is open) i can even connect to the r-pi but i always get "access denied" on any password i've set or even the defaults.

I think we had a similar issue to this a while ago, and the solution ended up being that you simply needed to wait a lot longer on the first boot of Raspberry Pi OS before trying to connect with SSH, as it needs to resize the filesystem, generate SSH keys, etc. etc.

tdewey-rpi, 2025-04-30:
Closing as:

1) Stale
2) Etcher???

closed 2025-04-30 by tdewey-rpi