raspberrypi/rpi-imager #1029: [BUG]: symbol(s) not found for architecture x86_64

k-schmidt, 2025-04-03:
### What happened?

Trying to build the RPI Imager on Mac OS Silicon. I replaced the line with where my QT installation is:
```
set(Qt6_ROOT "/Users/kyleschmidt/Qt/6.7.3/macos" CACHE PATH "Your Qt6 root path")
```
But I'm getting a build error. Logs attached.

### Version

1.9.2 (Default)

### What host operating system were you using?

macOS

### Host OS Version

15.3.2

### Selected OS

Haven't got that far yet

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
23:17:56: Running steps for project rpi-imager...
23:17:56: Starting: "/Users/kyleschmidt/Qt/Tools/CMake/CMake.app/Contents/bin/cmake" --build /Users/kyleschmidt/Downloads/rpi-imager-qml/src/build/Qt_6_7_3_for_macOS-Debug --target all
[  4%] Built target zlibstatic
[ 10%] Built target libzstd_static
[ 46%] Built target libcurl_static
[ 55%] Built target liblzma
[ 79%] Built target archive_static
[ 79%] Built target rpi-imager_autogen_timestamp_deps
[ 79%] Built target rpi-imager_autogen
[ 81%] Linking CXX executable rpi-imager.app/Contents/MacOS/rpi-imager
ld: warning: ignoring duplicate libraries: '-liconv'
ld: warning: ignoring duplicate libraries: '-liconv'
ld: warning: ignoring file '/opt/homebrew/Cellar/libnghttp2/1.65.0/lib/libnghttp2.dylib': found architecture 'arm64', required architecture 'x86_64'
Undefined symbols for architecture x86_64:
  "_nghttp2_http2_strerror", referenced from:
      _proxy_h2_on_stream_close in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _h2_handle_tunnel_close in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _on_stream_close in libcurl-d.a[x86_64][66](http2.c.o)
      _http2_handle_stream_close in libcurl-d.a[x86_64][66](http2.c.o)
      _http2_handle_stream_close in libcurl-d.a[x86_64][66](http2.c.o)
  "_nghttp2_is_fatal", referenced from:
      _cf_h2_proxy_send in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _proxy_h2_progress_egress in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _cf_h2_proxy_flush in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _h2_progress_egress in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      ...
  "_nghttp2_option_del", referenced from:
      _proxy_h2_client_new in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
[... 158 lines cut ...]
      _h2_submit in libcurl-d.a[x86_64][66](http2.c.o)
      _h2_submit in libcurl-d.a[x86_64][66](http2.c.o)
  "_nghttp2_submit_rst_stream", referenced from:
      _on_header in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      _on_stream_frame in libcurl-d.a[x86_64][66](http2.c.o)
      _http2_data_done in libcurl-d.a[x86_64][66](http2.c.o)
      _h2_xfer_write_resp in libcurl-d.a[x86_64][66](http2.c.o)
  "_nghttp2_submit_settings", referenced from:
      _cf_h2_proxy_ctx_init in libcurl-d.a[x86_64][11](cf-h2-proxy.c.o)
      _cf_h2_ctx_open in libcurl-d.a[x86_64][66](http2.c.o)
  "_nghttp2_submit_window_update", referenced from:
      _cf_h2_update_local_win in libcurl-d.a[x86_64][66](http2.c.o)
  "_nghttp2_version", referenced from:
      _Curl_http2_ver in libcurl-d.a[x86_64][66](http2.c.o)
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rpi-imager.app/Contents/MacOS/rpi-imager] Error 1
make[1]: *** [CMakeFiles/rpi-imager.dir/all] Error 2
make: *** [all] Error 2
23:17:57: The process "/Users/kyleschmidt/Qt/Tools/CMake/CMake.app/Contents/bin/cmake" exited with code 2.
23:17:57: Error while building/deploying project rpi-imager (kit: Qt 6.7.3 for macOS)
23:17:57: When executing step "Build"
```

lurch, 2025-04-03:
The critical line appears to be:
```
ld: warning: ignoring file '/opt/homebrew/Cellar/libnghttp2/1.65.0/lib/libnghttp2.dylib': found architecture 'arm64', required architecture 'x86_64'
```
I'm afraid I'm not very familiar with homebrew myself, but are you able to uninstall and then reinstall `libnghttp2` to see if it correctly installs the x86_64 version? If that doesn't work, perhaps there's a bug in homebrew's libnghttp2 package?

lurch, 2025-04-03:
Oh!! I misinterpreted from your original comment that you were on an x86_64 based Mac 🤦 

> I think the question is rather why is the Pi Imager build looking for x86_64 package architectures when it should know that arm64 is sufficient for my machine?

I dunno, but hopefully @tdewey-rpi will be able to provide a bit of support in amongst all the other projects that he's working on.
Could you please provide the exact command-lines that you've been using?

tdewey-rpi, 2025-04-03:
The problem here is that you're trying to build a universal binary, and you don't have nghttp2 available for both architectures.

If you're just looking to test locally, the simplest path forward is to adjust your CMake environment to only build for your host architecture: https://github.com/raspberrypi/rpi-imager/blob/ebd2a01c66237ce4a8c4ce3a5cdae6dcc307cc0f/src/CMakeLists.txt#L8

I expect a later version of rpi-imager to vendor nghttp2 to comprehensively address this.

tdewey-rpi, 2025-04-30:
I believe current `qml` should be OK for this - we now use FetchContent to vendor nghttp2 with our other dependencies, so you should be building a copy for both arm64 and x86_64 as part of a macOS build.

closed 2025-04-30 by tdewey-rpi